openwisp/openwisp-controller #344: [change] Improve system defined variables

nemesisdesign, 2020-12-11:
There are some minor issues with the system defined variables field which have to be addressed.

### Fallback for when no data is available

If no system variable is present, this is currently shown:

[image]

Not good. We need to show something like "There are no system defined variables available right now."

### Show/hide when height is higher than 150px

When the height of the system defined variables is greater than 150px, cut the content by limiting the height around 150px and add a **show** button.

Once the **show** button is clicked, the extra content is shown and **show** becomes **hide**, which works in the opposite way (hides the extra content and changes the wording to **show** again).

### Alphabetical ordering is not optimal

Alphabetical ordering is not optimal because system defined variables from VPN context uses letters that are higher up in the alphabet and hence make those variables show first, which are the least useful, because handled automatically by the system.

We should switch the `get_context` methods in Config, Template, Vpn to use `OrderedDict` in a way that ordering is retained, eg:

```
OrderedDict(
    (
        ('key1', 'val1'),
        ('key2', 'val2'),
    )
)
```

The right order to use for Device/Config is:

- name
- mac address
- id
- key
- all other keys after

For VPN is:

- ca
- cert
- key
- dh
- all other keys after

All the other dynamically generated context data should be ordered in alphabetical order but must be inserted after the most important data which is described above.

niteshsinha17, 2021-01-10:
I am working on it.

niteshsinha17, 2021-01-13:
> > I am working on it.
> 
> Thanks @niteshsinha17, do you confirm you're working on this? Assign it to yourself if you are.

Yes I am working on it. Completed its two task and working on 3rd.

niteshsinha17, 2021-01-15:
@nemesisdesign 
What should be the order of variables in the template?

nemesisdesign, 2021-01-15:
> What should be the order of variables in the template?

@niteshsinha17: alphabetical is good in that case, thanks.

PR #368 (2021-01-31): [change] Improved system defined variables #344
closes #344
commit fee4a057 (2021-01-31): [change] Improved system defined variables #344

Closes #344
closed 2021-01-31 by nemesisdesign