openwisp/openwisp-controller #353: [fix] Allow using variables also when config schema defines maxLength

nemesisdesign, 2020-12-20:
Whenever the config schema limits the max length of some fields (eg: ip address), variables cannot be used if their name exceed the `maxLength`, but this limitation is unintended: the contents of the variable should be checked, not the variable name itself.

This should be fixed both in the normal mode and the advanced mode.

codesankalp, 2021-03-16:
I will like to work on it and make a PR asap.

nemesisdesign, 2021-03-16:
The tricky part is that the generated field has a "maxlength" attribute set. Not sure how we can solve this.

@codesankalp thanks, read :point_up:

codesankalp, 2021-03-17:
@nemesisdesign I am trying to reproduce it but wasn't able to reproduce this issue. Can you help me with this?

nemesisdesign, 2021-03-18:
@codesankalp go to device page, ensure there's a configuration, scroll down, click on "configuration menu", then interfaces, then add an interface, then add an interface, add an ip address, try to use a variable in the IP address field, see that the maxlength attribute of the form field limits it? This is an issue for #402 because it prevents us from using variables there, which is super useful.

PR #410 (2021-05-20): [bug] Allow using variables also when config schema has maxLength  #353

Removing max_length attribute when the value starts with `{{...`.

Fixes #353
commit bca3d7b0 (2021-05-20): [fix] Allow using variables also when config schema has maxLength #353

Fixes #353
closed 2021-05-20 by nemesisdesign