openwisp/openwisp-controller #357: [bug] Shared templates can be edited by non superusers

nemesisdesign, 2020-12-30:
Seems shared templates (with no org defined) can be viewed and edited by non-superusers, which is wrong.

Something must have been changed which allowed shared templates to be seen by non superusers, we must fix it and add a test that ensures consistency.

Purhan, 2020-12-30:
I couldn't replicate this. Check the screenshots (dropdowns) attatched below.
<details><summary>Templates accessible by superuser</summary>
<p>
[image]
</p>
</details>
<details><summary>User `testuser1` with the following settings</summary>
<p>
[image]
</p>
</details>
<details><summary>Templates accessible by `testuser1`</summary>
<p>
[image]
</p>
</details>

nemesisdesign, 2020-12-30:
You're right @Purhan, must have been a mistake on my side, I tried again today with a fresher mind and also couldn't replicate it, thank you very much for confirming!

closed 2020-12-30 by nemesisdesign