openwisp/openwisp-controller #360: [feature] Add GeoJSON API endpoint with device location data

nemesisdesign, 2021-01-05:
We need to add a GeoJSON API endpoint (using [django-rest-framework-gis](https://github.com/openwisp/django-rest-framework-gis/)) which returns geographic data which can be used to show a general device map.

- the URL should be something like `/api/v1/device/geojson/`
- it should fetch all the devices accessible to the current logged in user: if it's superuser it can see everything, otherwise the user can see only objects related to organizations they manage
- we need to fetch locations which have devices deployed in them, and for each location we need to know how many devices there are, eg: `Location.objects.filter(devicelocation__isnull=False).annotate(device_count=Count('devicelocation'))`
- it should return at max 1000 items and paginate if more items are present, without breaking the GeoJSON format (see https://github.com/openwisp/django-rest-framework-gis/#geojsonpagination)
- each element should display all the information available in the fields of the `Location` model

ManishShah120, 2021-01-25:
I would like to take up this issue @nemesisdesign

Purhan, 2021-01-26:
@nemesisdesign Thanks for the detailed explanation. However, a few things are not clear to me:
- `/api/v1/device/geojson/` should display `devices` or `locations`? 
- Also since both `device` and `location` have their own organization field, which one should be the deciding factor?

nemesisdesign, 2021-01-26:
> @nemesisdesign Thanks for the detailed explanation. However, a few things are not clear to me:
> 
> * `/api/v1/device/geojson/` should display `devices` or `locations`?

we need to fetch locations which have devices deployed in them, and for each location we need to know how many devices there are, eg:

`Location.objects.filter(devicelocation__isnull=False).annotate(device_count=Count('devicelocation'))`

> * Also since both `device` and `location` have their own organization field, which one should be the deciding factor?

The two will always match, use the `organizaiton` field of the main model of the queryset, so if you follow the suggestion above, use the org field on the location model.

PR #374 (2021-02-18): [feature] Add geojson endpoints #360
Closes #360
commit 76325988 (2021-02-18): [feature] Added GeoJSON and location devices REST API endpoints #360

* [feature] Add geojson endpoints #360

Closes #360
commit 7b78fa04 (2021-02-18): [feature] Added GeoJSON and location devices REST API endpoints #360

Closes #360
closed 2021-02-18 by nemesisdesign