openwisp/openwisp-controller #361: [feature] Add a device name changed signal, connect it to refresh the notifications cache

nemesisdesign, 2021-01-07:
Blocked by https://github.com/openwisp/openwisp-notifications/issues/164.
We should first confirm that #164 can be done and then we can move on with this.

Like https://github.com/openwisp/openwisp-controller/issues/241 but emitted when the name is changed.

nemesisdesign, 2021-02-27:
**PS**: there's already a few lines of code which find out if the name has changed: https://github.com/openwisp/openwisp-controller/blob/a627b61ce149c95cf5d71d1a5b1e46de9e7bb373/openwisp_controller/config/base/device.py#L114-L115
We can change the implementation to be like https://github.com/openwisp/openwisp-controller/issues/241, which doesn't need an extra DB query which could be good to optimize the system a bit.

nemesisdesign, 2021-03-12:
This is ready to be worked on, depends on https://github.com/openwisp/openwisp-notifications/pull/168 which is ready, so that branch can be installed by editing the requirements.txt this way:

```
openwisp-notifications @ https://github.com/niteshsinha17/openwisp-notifications/tarball/issues/164-introduce-mechanism-to-clear-cache
```

codesankalp, 2021-03-12:
I would like to work on it.

PR #407 (2021-04-03): [feature] Added Device name changed signal #361
Added `device_name_changed` signal and connect it to refresh the notifications
cache.

Closes #361
commit 3b0e0ca1 (2021-04-03): [feature] Added Device name changed signal #361

Added `device_name_changed` signal.
Connected it to a receiver which refreshes the notifications cache.

Closes #361
closed 2021-04-03 by nemesisdesign