openwisp/openwisp-controller #367: [bug] Hostname validator should allow dots and underscore

nemesisdesign, 2021-01-16:
This is a [follow up to mailing list discussion about hostname validation](https://groups.google.com/g/openwisp/c/odjkLhfpQ7I/m/Vpvp8PcDDQAJ).

I have tested setting hostnames like `ap.master` and `ap_master` in OpenWRT and they are obviously allowed while OpenWISP rejects them. This is shouldn't happen.

niteshsinha17, 2021-01-27:
@nemesisdesign Can you please confirm `ap.master` is getting rejected. I am able to set this name. Only consecutive dots are not allowed but multiple dots are allowed. 
Also, I want to know this, can we have consecutive `underscores`? 
Thanks

nemesisdesign, 2021-01-28:
> @nemesisdesign Can you please confirm `ap.master` is getting rejected. I am able to set this name.

You're right! 

> Only consecutive dots are not allowed but multiple dots are allowed.

Ok :+1: 

> Also, I want to know this, can we have consecutive `underscores`?

Looking again at the existing literature, it's only allowed in linux but many definitions of hostnames do not allow underscores, so probably I was mistaken in opening this issue in the first place. Thank you for double checking @niteshsinha17! :+1:

closed 2021-01-28 by nemesisdesign