openwisp/openwisp-controller #370: [ssh] Decoding the output of the sysupgrade command fails with UnicodeDecodeError

okraits, 2021-01-21:
I did a firmware upgrade with openwisp-firmware-upgrader. It was successful, but I got the following exception twice:

    Traceback (most recent call last):
      File "/home/okraitschy/owm_new/build/venv/lib/python3.7/site-packages/billiard/process.py", line 327, in _bootstrap        
        self.run()                                         
      File "/home/okraitschy/owm_new/build/venv/lib/python3.7/site-packages/billiard/process.py", line 114, in run                
        self._target(*self._args, **self._kwargs)
      File "/home/okraitschy/owm_new/build/venv/lib/python3.7/site-packages/openwisp_firmware_upgrader/upgraders/openwrt.py", line
     134, in upgrade                                                                                                             
        conn.exec_command(command, timeout=timeout)                                                                              
      File "/home/okraitschy/owm_new/build/venv/lib/python3.7/site-packages/openwisp_controller/connection/connectors/ssh.py", line 136, in exec_command
        output = output_str.decode('utf8').strip()  
    UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc3 in position 6586: invalid continuation byte

Due to `sysupgrade -v`, the verbose output of sysupgrade is decoded and that fails because of some byte.

[https://github.com/openwisp/openwisp-controller/blob/master/openwisp_controller/connection/connectors/ssh.py#L136](url)

Just some thoughts:

Why do we print out the output?
Why do we decode it?
Why don't we catch exceptions there?

nemesisdesign, 2021-01-21:
@okraits:

> Why do we print out the output?

if I remember correctly, `print` is used for logging and could be changed to `logger.info()`

> Why do we decode it?

I don't remember now, it may be that the returned value is `byte`. Not sure.

> Why don't we catch exceptions there?

I didn't account for this failure.

okraits, 2021-01-21:
> if I remember correctly, `print` is used for logging and could be changed to `logger.info()`

Yeah, I was also thinking about converting this to logging.

> I don't remember now, it may be that the returned value is `byte`. Not sure.

I will have a look into it.

> I didn't account for this failure.

I didn't mean that as an accusation. Thank you for your great work!

I'll try to come up with a PR for this.

PR #372 (2021-01-25): [fix] Connection: Improve logging and unicode conversion #370
Fixes #370

Another issue related to `openwisp-firmware-upgrader` is that `/sbin/sysupgrade` calls `ubus call system sysupgrade` in the end, which can exit with != 0 even if the sysupgrade is successful (for me a successful sysupgrade returns -1).
commit 24cc1a07 (2021-01-25): [fix] Connection: ignore unicode conversion issues, improved logging #370

Fixes #370
commit e36619a6 (2021-04-09): [fix] Connection: ignore unicode conversion issues, improved logging #370

Fixes #370
closed 2021-01-25 by nemesisdesign