openwisp/openwisp-controller #377: [extending] openwisp-users cannot be extended.

atb00ker, 2021-02-01:
The openwisp-users models are hardcoded in places, example:
- [admin file with hardcoded model](https://github.com/openwisp/openwisp-controller/blob/master/openwisp_controller/config/admin.py#L27)

To solve this problem, we should:
- Use swapper
- [Test by extending openwisp-users like done in openwisp-radius](https://github.com/openwisp/openwisp-radius/tree/master/tests/openwisp2)

nemesisdesign, 2021-02-01:
@atb00ker I'd be happy also with simply fixing the import, without extending openwisp_users, although the test would surely allow us to ensure everything works but if we follow that approach then we should do it also in firmware-upgrader with the controller models and in monitoring with the controller models, but we risk to bloat the test code and make our life painful so we should be cautious.

atb00ker, 2021-02-05:
@nemesisdesign I understand but if we are providing the feature to extend, I think it's easier to develop it when we can test by extending as well, but I understand your point too! :smile:

okraits, 2021-02-05:
I think it should be verified and tested that the extension feature actually works and that all modules are prepared.

After updating our setup to the latest releases of all OpenWISP modules I started to create extended apps for all apps of openwisp-controller and openwisp-users, investing many hours to get this to work until I realized that there are still issues like this one (and possibly others). I noticed I was probably the first one to really use this feature and then resorted to add my changes with few patches.

nemesisdesign, 2021-02-05:
@atb00ker @atb00ker when important issues like this arise, please help me to make sure these are listed in the priority board.
The same issue is affecting other modules, I just did a project wide search on all the openwisp modules.
OpenWISP uses swapping with django-organizations and django-notifications-hq successfully, but we did the effort to patch those modules first, so the same process will have to be repeated with this new feature.
I'm going to open issues in all the modules I find and report.

nemesisdesign, 2021-02-05:
IPAM suffers the same issue.
Docker-openwisp has several imports in which models are imported directly.
Network topology only had a couple of lines in the tests: https://github.com/openwisp/openwisp-network-topology/commit/c275a5a609b1a64ea12054870e05861b08a0a237
The most important module suffering the issue is this one.

commit 24aadc09 (2021-02-18): [extending] Extending openwisp-users #377

Closes #377
commit 63a81052 (2021-02-24): [extending] Extending openwisp-users #377

Closes #377
commit 220e45df (2021-03-11): [extending] Extending openwisp-users #377

Closes #377
commit 960caa83 (2021-03-15): [extending] Extending openwisp-users #377

Closes #377
closed 2021-04-03 by nemesisdesign