openwisp/openwisp-controller #379: [feature] REST API for main controller features

nemesisdesign, 2021-02-04:
We need to implement a RESTful HTTP API for openwisp-controller, the API will allow to perform the following CRUD (Create, Retrieve, Update, Destroy) operations on the following objects:

- Device (including configuration details, assigned templates, and other points explained below) `/api/v1/controller/device/`
- Template: `/api/v1/controller/template/`
- VPN Server: `/api/v1/controller/device/`

Important notes about the Device API:

- The device API should provide not only the device data but also the configuration data (if present, please handle the case in which the Config object is not available) and templates.
- It should allow assigning/unassigning and changing the order of templates.
- It should take into account the new required templates feature, because trying to remove a required template will raise an exception, so the code shall catch this exception and return an error message gracefully.
  The required templates feature is at https://github.com/openwisp/openwisp-controller/pull/340, this means that we would branch off that feature branch.
- There should be an additional endpoint through which the configuration can be downloaded, eg:
  - `GET /api/v1/controller/device/{pk}/configuration/`
  - `GET /api/v1/controller/template/{pk}/configuration/`
  - `GET /api/v1/controller/vpn/{pk}/configuration/`
- The API endpoints must be paginated and respect the django permission system (DjangoPermission DRF class)
- They must be available to authenticated users only and respect multi-tenancy, we already have some work for this done in openwisp-users https://github.com/openwisp/openwisp-users#django-rest-framework-permission-classes (Please read the DRF permission classes and DRF mixins sections).

We also need to make sure the auto generated API documentation based on  OpenAPI (also known as swagger, we use drf_yasg), as well as basic documentation in prose added to the README of openwisp-controller (please see how we did it in openwisp-firmware-upgrader).

We need tests (add a new file `test_api` and basic docs like we have in firmware-upgrader, let's just copy the structure of the REST API docs of firmware-upgrader: https://github.com/openwisp/openwisp-firmware-upgrader/#rest-api.

ManishShah120, 2021-02-07:
I would love to work on this @nemesisdesign, Can I be assigned on this?

ManishShah120, 2021-02-13:
> * Device (including configuration details, assigned templates, and other points explained below) `/api/v1/controller/device/`
> * Template: `/api/v1/controller/template/`
> * VPN Server: `/api/v1/controller/device/`

@nemesisdesign for the VPN Server shouldn't the endpoint be as `/api/v1/vpn/` :thinking:

nemesisdesign, 2021-02-15:
> > * Device (including configuration details, assigned templates, and other points explained below) `/api/v1/controller/device/`
> > * Template: `/api/v1/controller/template/`
> > * VPN Server: `/api/v1/controller/device/`
> 
> @nemesisdesign for the VPN Server shouldn't the endpoint be as `/api/v1/vpn/`

@ManishShah120 correct, it's a typo, thanks for letting me know, I corrected the description.

ManishShah120, 2021-02-16:
@nemesisdesign , @atb00ker `template`, and `vpn` part of the API is done, but am struggling a bit figuring out a way to fulfill the extra requirements mentioned for the `device` endpoint. Can you please have a look at the implementation till now and also if there are any references I should have a look at for getting done with the `device` endpoint please let me know.
--Thank You

nemesisdesign, 2021-02-17:
> @nemesisdesign , @atb00ker `template`, and `vpn` part of the API is done, but am struggling a bit figuring out a way to fulfill the extra requirements mentioned for the `device` endpoint. Can you please have a look at the implementation so far and also if there are any references I should have a look at for getting done with the `device` endpoint please let me know.
> _Once this gets done I'll start writing the tests and document them in the README._
> --Thank You

@ManishShah120 please open a draft PR.

nemesisdesign, 2021-02-24:
Leaving a note here regarding discussions done via IM:

a PUT request to the device detail endpoint should be able to update all the fields that are editable in the django admin
regarding the config templates, the PUT request to the device detail page should be able to update the order field of each m2m relation, effectively changing the order of the templates. For the PUT request to add/remove templates, the JSON should just add new templates or remove templates that have to be removed.

PR #386 (2021-05-12): [feature] REST API for main controller features #379
Closes #379
commit 91084747 (2021-05-12): [feature] REST API for main controller features #379

The history of the changes and discussion is available at
https://github.com/openwisp/openwisp-controller/pull/386

Closes #379
closed 2021-05-12 by nemesisdesign