openwisp/openwisp-controller #385: [bug] Floorplan organization can be different than location organization

Purhan, 2021-02-16:
While registering a new floorplan, there is no choice to select the organization, which infers that it should be the same as the location.

After creating a location (Let's say `location_a` for `organization_a`), if we create a floorplan, its organization will also be `organization_a`

However, if the organization of location is changed (for example, to `organization_b`) the floorplan will still have the old organization

nemesisdesign, 2021-03-25:
Thanks for opening this @Purhan. I think we could assess whether it makes sense to have the organization field on the floorplan or not, probably not, if we can avoid it and modify the code accordingly (also avoiding to generate too many queries) we could solve it this way.

oceankumar, 2026-10-13:
I reproduced this on current master (0f0de551): after saving a location in another organization, its floorplan still belongs to the old organization. The existing 12 geo model tests and repository QA pass; a regression asserting the floorplan's new organization fails.

I'd like to continue the model-level approach from the author-closed #1210, incorporating the review's swappable-model guidance. I plan to synchronize only that location's floorplans when its organization is persisted, preserving partial-save behavior and other locations. I'll cover multiple floorplans, unchanged organizations, update_fields and the swapped sample app.

Please let me know if this misses a requirement.