openwisp/openwisp-controller #399: [bug] Sortedm2m causes vpnclients to be destroyed and recreated every time the configuration or templates are changed

nemesisdesign, 2021-03-05:
**How to replicate:**

- create a device
- create a configuration, assign a template of type "VPN client" which has auto-cert enabled, save
- go to the certificates list, you'll see a new certificate has been created
- change the details of the configuration (eg: the context or the config), or alternatively, add a new template

**Expected result:**

The certificate created in the previous steps should still exists.

**Actual result:**

The certificate is destroyed and a new one is created.

This is caused by the annoying implementation of django-sortedm2m, which implements the reodreding of m2m relationships by clearing templates and adding them back again, which causes signal receivers of openwisp to destroy and recreate many different things.

I'm not sure if we can solve it without forking django-sortedm2m and implement the reordering in a different way, nor I'm sure if it's worth pursuing this now, but I want to leave this here for others to know so at least we can be aware of it.

nemesisdesign, 2021-03-25:
I thought that we could solve it in a different way:

- avoid clearing vpnclients on post_clear
- when creating new vpnclients, check if they already exists for that config and vpn, if they do, skip

Saurav-Shrivastav, 2021-03-25:
I will start working on this and make a PR soon.

nemesisdesign, 2021-03-25:
@devkapilbansal good question, but there's no concept of ordering in VPN clients, so template m2m relationships (config.templates) have to be deleted and recreated, because those one do have the concept of order, what we have to avoid being deleted and recreted are VpnClient instances, VpnClient is a custom m2m through model.

So, templates through emit `m2m_changed` when they're changed, we connect a receiver to this signal:
https://github.com/openwisp/openwisp-controller/blob/32c4bc9bec9dcefc24ca2498765ed4253b0a3b03/openwisp_controller/config/apps.py#L62-L66

The receiver function (actually a class method) is here:
https://github.com/openwisp/openwisp-controller/blob/32c4bc9bec9dcefc24ca2498765ed4253b0a3b03/openwisp_controller/config/base/config.py#L207-L241

To anyone wanting to work on this: please write a test first, eg:

- create a vpn server
- create a vpn client template
- create a new device and config and assign it the template created at the previous step
- ensure the vpnclient instance is created and with it the x509 `Cert` instance
- now call device.config.templates.clear()
- ensure the related `VpnClient` and `Cert` are not deleted
- now assign the template to the config object again and ensure no additional VpnClient is created and everything runs fine

Please do not forget updating the docs.

Saurav-Shrivastav, 2021-03-30:
@nemesisdesign Very naive question - Where should the test for the solution ideally go?

nemesisdesign, 2021-03-30:
> @nemesisdesign Very naive question - Which file should be ideally used for writing the test for this solution?

@Saurav-Shrivastav take a look at https://github.com/openwisp/openwisp-controller/blob/master/openwisp_controller/config/tests/test_config.py#L297-L379.

There's also one test which will have to be changed (the one which tests the `clear()` operation).

PR #412 (2021-04-05): [fix] VpnClients destroyed on any change in configuration or template…
…s fixed #399

- Modified `manage_vpn_clients` callback method
- vpnclients are not cleared on post_clear
- for a post_add action, a new vpnclient is created only if it doesn't exist already
- added a Test for the above functionality
- modified 2 tests (viz - `test_clear_vpnclients` and `test_automatically_created_cert_deleted_post_clear`) to incorporate the new functionality

Fixes #399
commit a2fc418f (2021-04-05): [fix] Do not destroy VpnClient instances on post_clear #399

Fixes #399
commit 4e2ffba6 (2021-04-09): [fix] Do not destroy VpnClient instances on post_clear #399

Fixes #399
commit 1c17128e (2021-04-09): [fix] Do not destroy VpnClient instances on post_clear #399

Fixes #399
closed 2021-04-05 by nemesisdesign