openwisp/openwisp-controller #400: [feature] Automatic provisioning of Subnets and IPs

pandafy, 2021-03-06:
Using subnet subdivision rules created by the user, automatically provision subnets and IPs for the Device.

### Requirements

The `SubnetSubdivisionRule` will have the following fields: 

1. Organization
2. Reference label - an alphanumeric label to help users reference this rule
3. Size of generated subnet 
4. Amount of subnets 
5. Amount of IPs 

The generated subnets and IPs should be able to reference using variables in configurations(`Config`).  

**Example**: 

A `SubnetSubdivisionRule` having the following fields

```
Reference label: OP
Amount of subnets: 1
Size: /24
Amount of IP addresses: 1
```

should generate the following artifacts:

- A subnet `10.0.0.0/24` accessible by `OP_subnet1` variable
- An IP `10.0.0.1` accessible by `OP_subnet1_ip1` variable
- Additionally `OP_prefixlen` should resolve to `24` (the size of generated subnet)

#### Additional Requirements

Add elements to the admin page of Device and Subnet to allow users to quickly check related devices and subnets.  

### Possible Implementation

Add a new submodule that will bridge between openwisp-ipam and controller. 

The submodule will add the following to the device admin of openwisp-controller:
- [a filter in the device admin that allows inserting text](https://hakibenita.com/how-to-add-a-text-filter-to-django-admin) (showing all the subnets will destroy the UI and slow down the UI generation), the text will contain the subnet

The submodule should add the following to the openwisp-ipam admin:
- a link in the subnet edit page which redirects to the filtered device list page. (A user should be able to see which device uses the subnet they are currently editing)
- a new column in the subnet list page with the same link(as above), so that each element in the subnet list will have a link to go to the related devices 

#### Notes

- Maintain an additional database table that will serve as an index to quickly resolve variables(e.g. `OP_subnet1_ip1`). The index will have the following details:
    - Variable Name
    - `SubnetSubdivisionRule`
    - `Subnet`
    - `IPAddress`
    - `Config` (helpful in rendering related objects in Django admin)

____________________________________________

**Note:** This will introduce `openwisp-ipam` as a direct dependency of openiwsp-controller

okraits, 2021-03-06:
Could you please explain the actual usecase? Do you want to use that to assign subnets for OpenVPN?

nemesisdesign, 2021-03-07:
The use case is to allow users to automatically generate subnets and ips according to their rules, then use these in templates to implement their routing.

pandafy, 2021-04-12:
TODO: Update document links before merging.

commit c5bfdc98 (2021-04-15): [feature] Added automatic provisioning of Subnets and IPs #400

- Added subnet division rules for defining charecterstic of
  provisioned subnets and IPs
- Added "VPN" rule type that provisions subents and IPs for a
  device when a new VpnClient is created
- Added support for adding custom subnet division rule types
- Added support for updating subnet divisioon rules
  Currently only "number_of__ips" and "label" fields can be updated
- Added an inline admin for subnet division rules in SubnetAdmin
- Added a filter on SubnetAdmin to allow filtering by related device name
- Added a filter on Dev [... cut]
PR #402 (2021-04-15): [feature] Automatic provisioning of subnets and IPs 
Closes #400
commit 1005efa2 (2021-04-15): [feature] Added automatic provisioning of Subnets and IPs #400

- Added subnet division rules for defining charecterstic of
  provisioned subnets and IPs
- Added "VPN" rule type that provisions subents and IPs for a
  device when a new VpnClient is created
- Added support for adding custom subnet division rule types
- Added support for updating subnet divisioon rules
  Currently only "number_of__ips" and "label" fields can be updated
- Added an inline admin for subnet division rules in SubnetAdmin
- Added a filter on SubnetAdmin to allow filtering by related device name
- Added a filter on Dev [... cut]
commit 597f107f (2021-05-07): [feature] Added automatic provisioning of Subnets and IPs #400

- Added subnet division rules for defining charecterstic of
  provisioned subnets and IPs
- Added "VPN" rule type that provisions subents and IPs for a
  device when a new VpnClient is created
- Added support for adding custom subnet division rule types
- Added support for updating subnet divisioon rules
  Currently only "number_of__ips" and "label" fields can be updated
- Added an inline admin for subnet division rules in SubnetAdmin
- Added a filter on SubnetAdmin to allow filtering by related device name
- Added a filter on Dev [... cut]
closed 2022-01-19 by nemesisdesign