openwisp/openwisp-controller #417: [ci] CI not testing on django~=3.0.0

pandafy, 2021-04-05:
There is a mistake in defining django version in CI 

```diff
--- a/.github/workflows/ci.yml
+++ b/.github/workflows/ci.yml
@@ -29,8 +29,8 @@ jobs:
           - 3.6
           - 3.7
         django-version:
-          - django~=3.0
-          - django~=3.1
+          - django~=3.0.0
+          - django~=3.1.0
 
     steps:
     - uses: actions/checkout@v2
```

The CI build is always using django~=3.1.0. Found this while working on #402. I wonder what made [this build to fail for django 3.0.0](https://github.com/openwisp/openwisp-controller/pull/402/checks?check_run_id=2264998129). Thanks to whatever it was, it made me dive deeper into build logs. 

I opened this issue instead of fixing it right away because some tests in sample apps were also failing after making such change. It will require further debugging and fixing.

pandafy, 2021-04-05:
```diff
--- a/tests/openwisp2/sample_users/migrations/0001_initial.py
+++ b/tests/openwisp2/sample_users/migrations/0001_initial.py
@@ -21,9 +21,11 @@ class Migration(migrations.Migration):
     initial = True
 
     dependencies = [
-        ('auth', '0012_alter_user_first_name_max_length'),
+        ('auth', '0011_update_proxy_permissions'),
     ]
```

`0012_alter_user_first_name_max_length` migration file is not available in django~=3.0.0

codesankalp, 2021-04-05:
@pandafy I will try to fix this.

devkapilbansal, 2021-04-05:
> ```diff
> --- a/tests/openwisp2/sample_users/migrations/0001_initial.py
> +++ b/tests/openwisp2/sample_users/migrations/0001_initial.py
> @@ -21,9 +21,11 @@ class Migration(migrations.Migration): [...]

Actually there should not be any hard dependencies like this

nemesisdesign, 2021-04-07:
So this issue is effectively affecting all the builds :facepalm: :cry: 
@codesankalp @devkapilbansal can you help to fix it in all the CI builds which you worked on please?

PR #418 (2021-04-07): [ci] CI fixed for testing on django~=3.0.0 #417
Changed the version of Django to use the exact version in the CI build.

Fixes #417
commit 31044e03 (2021-04-07): [ci] CI fixed for testing on django~=3.0.0 #417

Fixes #417
commit 94177296 (2021-04-09): [ci] CI fixed for testing on django~=3.0.0 #417

Fixes #417
commit 6f8bab4f (2021-04-09): [ci] CI fixed for testing on django~=3.0.0 #417

Fixes #417
closed 2021-04-07 by nemesisdesign