openwisp/openwisp-controller #437: [bug] Devices don't apply downloaded configuration

apoulias, 2021-04-23:
Some of my devices don't get the updated configuration applied after a template is changed. The configuration archive is downloaded correctly in /tmp/openwisp but doesn't get applied. The behavior seems random so I can't exactly describe the steps to reproduce. 

My setup has three devices with a management VPN and two templates. One template holds some variables and the other the wifi configuration. I'm doing frequent changes (one per minute) to the configuration template and check if they are applied correctly. When the status of all the devices turn to applied (some times even it actually isn't), I repeat.

Looking at the logs of Openwrt (in the failed device) it is noticeable that the ssh connection from the controller happens twice.
Additionally, looking at the logs of celery in the controller it seems that the update_template_related_config_status is also called twice and there are four authentications instead of three. 

Regarding the Openwrt part, my guess is that the downloaded configuration isn't applied because of the second ssh connection from the controller.

I've also noticed the behavior (multiple update_template_related_config_status) to happen more than twice, but it's more rare.

Please correct me if I'm wrong in my observations/guesses..

I'm using openwisp-controller 0.8.4 and openwisp-config 0.5.0.

Attached are both the logs of Openwrt and the controller.
[openwrt_bug_log.txt](https://github.com/openwisp/openwisp-controller/files/6365881/openwrt_bug_log.txt)
[controller_log.txt](https://github.com/openwisp/openwisp-controller/files/6365882/controller_log.txt)

apoulias, 2021-05-04:
It seems that the issue is related to redis. Check that [issue](https://github.com/celery/celery/issues/6358). I'm currently running a RabbitMQ broker just for Celery and it works fine.

nemesisdesign, 2021-05-05:
Thanks @apoulias for adding info! :+1: 

The issue linked mentions retries, but if I'm not mistaken openwisp-controller does not use retries.

With this change in openwsip-config (https://github.com/openwisp/openwisp-config/commit/7cbb2eb2f07a6bdbff87c8034d7bf22efc510c6e), in the near future we'll be able to reload the config without restarting the agent which is a lot better.

However this bug is a different beast. I have reasons to believe it is possible that two SSH connections are initiated by OpenWISP because it happened in the past, I thought I had solved it, there's even tests for it but simulated/mocked tests are not 100% guaranteed to work in the real world so I will have to double check when I can. 

So I suggest keeping an eye on this and let us know if you incur in this issue again.
I will surely have a chance to double check although it may take me some time.

apoulias, 2021-05-06:
@nemesisdesign I'll keep an eye on that. 

Regarding the multiple SSH connections, at least in my case, I'm pretty convinced it happened because of the multiple simultaneous tasks received by Celery.

As soon as the Save button on the template was pressed, multiple "update_template_related_config_status" tasks were received by Celery. After making sure that the save method of template (add a log entry and was called once) and the transaction.on_commit (replaced it with direct call of the task) don't cause the problem, the only suspect was Celery. After trying different settings (e.g. visibility_timeout) which didn't solve the issue, I switched to RabbitMQ and if was fixed.

Maybe it's a completely different case than the one you have in mind. Anyways, I'll be glad to help if you have any suggestions.

nemesisdesign, 2022-01-25:
I was not able to replicate this issue and hence I am closing it, if anyone can replicate this issue please reopen and provide more detailed information.

closed 2022-01-25 by nemesisdesign