openwisp/openwisp-controller #439: [bug] Popup never closes

atb00ker, 2021-04-23:
When I create a new connection credentials successfully, the popup never closes it gets stuck on "Closing Popup..."
Steps to reproduce:

1. Go to a device without connection
2. Press `+` to create a new connection
3. Fill correct details to fill the credential
4. Hit save and notice while credentials are saved, popup remains open.

yashikajotwani12, 2022-02-02:
I would like to work on this.

atb00ker, 2022-02-04:
Hey @yashikajotwani12 ,

Feel free to pick it, no assignment or permission required! :smile:

saieswar13, 2022-02-06:
Is this issue still open, can I work on this issue?

yashikajotwani12, 2022-02-07:
> Is this issue still open, can I work on this issue?

Hey @saieswar13, I am working on this, can you take another issue, thanks.

yashikajotwani12, 2022-02-15:
Before: 
[image]

After:
[image]

Hello! I have made changes in virtual env's file widget.js 
```
 // var name = windowname_to_id(win.name);
                var elem = $('#' + win.name);
```
so like how I should change it actually to make it work like this?

commit dd6fd95d (2022-02-16): [deps] Upgraded dependencies #439

- Bumped django-taggit~=2.1.0
- Bumped scp~=0.14.2
- Bumped django-sortedm2m~=3.1.1

Test dependencies:
- pytest-django~=4.5.2
- Bumped redis~=4.1.2
- Bumped mock-ssh-server~=0.9.1

Removed coverage dependency because it's defined in openwisp-utils.

Closes #439