openwisp/openwisp-controller #459: [API][config] Functional testing results

atb00ker, 2021-05-15:
Legends:
[SU] - Tested with Superuser
[O1A] - Tested with Org1 Administrator
[O1O] - Tested with Org1 Operator
Status:Pass - Behaves like admin UI or deviation documented
Status:Fail - Does not behave like admin UI
Status:NTD - Need to discuss

atb00ker, 2021-05-15:
| SN |                         /api/v1/controller/template/                         | Status |                                                      Details                                                      |
|:--:|:----------------------------------------------------------------------------:|:------:|:-----------------------------------------------------------------------------------------------------------------:|
| 1  | [SU] Can view all templates                                                  | Pass   |                                                                                                                   |
| 2  | [O1A] can view only org1 templates (not even shared org)                     | Pass   |                                                                                                                   |
| 3  | [O1O] can view no templates                                                  | Pass   |                                                                                                                   |
| 4  | [SU][O1A] Field validation error on field                                    | Pass   |                                                                                                                   |
| 5  | [SU][O1A] All errors reported together                                       | Fail   | ([Check Image 1](https://ibb.co/1G3JM5C)) This error was reported after I fixed other errors and send request.    |
| 5  | [SU][O1A] Allows creating template without config                            | Fail   | Cannot create template without config (deviation from Admin UI)                                                   |
| 6  | [SU][O1A] Allows creating template with config                               | Pass   |                                                                                                                   |
| 7  | [SU][O1A] Allow creating VPN template                                        | Pass   |                                                                                                                   |
| 8  | [SU][O1A] Allow creating VPN template without openvpn section                | NTD    | Section automatically added in Admin UI. Is this acceptable deviation?                                            |
| 9  | [O1O] Access denied to org1 operator user                                    | NTD    | Currently, operator can see the page but can't view / edit / create templates, should we restrict access to page? |
| 10 | [SU] Unique template name ensured                                            | Pass   |                                                                                                                   |
| 11 | [O1A] Trying to create template for another org gives object not found error | Pass   |                                                                                                                   |
|    |                                                                              |        |                                                                                                                   |
|    |                                                                              |        |                                                                                                                   |

atb00ker, 2021-05-15:
| SN |                       controller/template/str:pk/                       | Status | Details |
|----|:-----------------------------------------------------------------------:|:------:|---------|
|  1 | [SU] Can view any templates                                             |  Pass  |         |
|  2 | [O1A] Can view only org1 templates (not even shared org)                |  Pass  |         |
|  3 | [O1O] Can view no templates                                             |  Pass  |         |
|  4 | [SU][O1A] Correct data of the template shown                            |  Pass  |         |
|  5 | [SU][O1A][O1O] Can only delete templates if they have correct privilege |  Pass  |         |

atb00ker, 2021-05-15:
| SN |        controller/template/str:pk/configuration       | Status |                               Details                               |
|:--:|:-----------------------------------------------------:|:------:|:-------------------------------------------------------------------:|
|  1 |     [SU][O1A][O1O]  Can download any configuration    |   NTD  | Should any auth user, even from another org be allowed to download? |
|  2 | Non-authenticated user cannot download configurations |  Pass  |                                                                     |
|  3 | [SU][O1A] Correct template config downloaded          |  Pass  |                                                                     |

atb00ker, 2021-05-15:
| SN |                                  controller/vpn/                                 | Status |                                      Details                                     |
|:--:|:--------------------------------------------------------------------------------:|:------:|:--------------------------------------------------------------------------------:|
|  1 | [SU] Can view any VPN                                                            |  Pass  |                                                                                  |
|  2 | [O1A] Can view only org1 VPN (not even shared org)                               |  Pass  |                                                                                  |
|  3 | [O1O] Cannot view any VPN                                                        |  Pass  |                                                                                  |
|  4 | [SU][O1A] Field validation error on field                                        |  Pass  |                                                                                  |
|  5 | [SU][O1A] All errors reported in one go                                          |  Fail  | If name, host, key, ca, config are all null, config error is not shown.          |
|  6 | [SU] Only valid config schema accepted                                           |  Pass  |                                                                                  |
|  7 | [SU][O1A] Can create VPN                                                         |  Pass  |                                                                                  |
|  8 | [SU][O1A] Cannot create VPN                                                      |  Pass  |                                                                                  |
|  9 | [SU][O1A] Unique name per org check                                              |  Pass  |                                                                                  |
| 10 | [SU][O1A] Appropriate CA/Certs visible                                           |  Fail  | ([See image 2](https://i.imgur.com/hXX03Ut.png))O1A does not see shared CA/Certs |
| 11 | [01A] Trying to create device with another org's id gives object not found error |  Pass  |                                                                                  |

ManishShah120, 2021-05-15:
> SN	/api/v1/controller/template/	Status	Details
> 5	[SU][O1A] All errors reported together	Fail	([Check Image 1](https://ibb.co/1G3JM5C)) This error was reported after I fixed other errors and send request.
> 6	[SU][O1A] Allows creating template without config	Fail	Cannot create template without config (deviation from Admin UI)

@atb00ker ragarding SN5 & SN6 when the template type is generic the admin panel also doesn't allow us to create template without config.

atb00ker, 2021-05-15:
| SN |                          controller/vpn/str:pk/                         | Status |   Details   |
|:--:|:-----------------------------------------------------------------------:|:------:|:-----------:|
| 1  | [SU] Can view any vpn                                                   | Pass   |             |
| 2  | [O1A] Can view only org1 vpns (not even shared org)                     | Pass   |             |
| 3  | [O1O] Cannot view any vpn                                               | Pass   |             |
| 4  | [SU][O1A] Correct data of the vpn shown                                 | Pass   |             |
| 5  | [SU][O1A][O1O] Can only delete templates if they have correct privilege | Pass   |             |

atb00ker, 2021-05-15:
| SN |                         Status                        | Status |                               Details                               |
|:--:|:-----------------------------------------------------:|:------:|:-------------------------------------------------------------------:|
|  1 |     [SU][O1A][O1O]  Can download any configuration    |   NTD  | Should any auth user, even from another org be allowed to download? |
|  2 | Non-authenticated user cannot download configurations |  Pass  |                                                                     |
|  3 |      [SU][O1A] Correct template config downloaded     |  Pass  |                                                                     |

atb00ker, 2021-05-15:
| SN |                          /api/v1/controller/device/                          | Status |                                                                                                                          Details                                                                                                                         |
|:--:|:----------------------------------------------------------------------------:|:------:|:--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------:|
|  1 |                          [SU] Can view all templates                         |  Pass  |                                                                                                                                                                                                                                                          |
|  2 |           [O1A] can view only org1 templates (not even shared org)           |  Pass  |                                                                                                                                                                                                                                                          |
|  3 |                          [O1O] can view no templates                         |  Pass  |                                                                                                                                                                                                                                                          |
|  4 |                     [SU] Field validation error on field                     |  Pass  |                                                                                                                                                                                                                                                          |
|  5 |                    [SU][O1A] All errors reported together                    |  Pass  |                                                                      ([Check Image 1](https://ibb.co/1G3JM5C)) This error was reported after I fixed other errors and send request.                                                                      |
|  6 |                       [SU][O1A] Allows creating device                       |  Pass  |                                                                                                                                                                                                                                                          |
|  8 |                     [SU][O1A] Allow creating VPN template                    |  Pass  |                                                                                                                                                                                                                                                          |
|  9 |                   [O1O] Access denied to org1 operator user                  |  Fail  | DRF doesn't implement the view [permission by default](https://www.django-rest-framework.org/api-guide/permissions/#djangomodelpermissions). [We should help handle it](https://github.com/encode/django-rest-framework/pull/6325#issuecomment-8401133). |
| 10 |                 [SU] Unique device name / mac-address ensured                |  Pass  |                                                                                                                                                                                                                                                          |
| 11 | [O1A] Trying to create template for another org gives object not found error |  Pass  |                                                                                                                                                                                                                                                          |

atb00ker, 2021-05-15:
SN | controller/device/str:pk/ | Status | Details
-- | -- | -- | --
1 | [SU] Can view any device | Pass |  
2 | [O1A] Can view only org1 device | Pass |  
3 | [O1O] Cannot view any device | Pass |  
4 | [SU][O1A] Correct data of the device shown | Pass |  
5 | [SU][O1A][O1O] Can only delete templates if they have correct privilege | Pass |

atb00ker, 2021-05-15:
SN | Status | Status | Details
-- | -- | -- | --
1 | [SU][O1A][O1O]  Can download any configuration | NTD | Should any auth user, even from another org be allowed to download?
2 | Non-authenticated user cannot download configurations | Pass |  
3 | [SU][O1A] Correct template config downloaded | Pass |

nemesisdesign, 2021-12-03:
@atb00ker I guess we can close this right?

atb00ker, 2021-12-07:
Yep, yep! Closing! 😄

closed 2021-12-07 by atb00ker