openwisp/openwisp-controller #460: [API] Change device location API endpoint again

nemesisdesign, 2021-05-17:
We should change this again to make it consistent: `/device/{id}/location` > `​/controller​/device​/{id}/location/`.

We should maintain the old URL for backward compatibility.

@atb00ker @ManishShah120 is there a way to avoid showing the old URL in the API docs but keeping it around for backward compatibility until the next version?

atb00ker, 2021-05-17:
Sounds good! 
> is there a way to avoid showing the old URL in the API docs but keeping it around for backward compatibility until the next version?

Will need to check if swagger has something like that! 😄

ManishShah120, 2021-06-04:
> PS: I also noticed we have some URLs prefixed with `location`, to maintain consistency we should name these:
> 
> * `/controller/location/`

Hi, @nemesisdesign Did you mean to change the endpoints name to `controller_location`

atb00ker, 2021-06-04:
I think he means change `<host>/location/` -> `<host>/controller/location/`.

commit 7ab23c3e (2021-06-04): [change] Changed API endpoints of geo app #460

Closes #460
commit 1621a8e6 (2021-06-09): [change!] Changed API endpoints of geo app #460

Closes #460
PR #483 (2021-06-09): [change] Changed API endpoints of geo app #460
Closes #460
commit 5d14070a (2021-06-09): [change!] Changed API endpoints of geo app #460

Closes #460
closed 2021-06-09 by nemesisdesign