openwisp/openwisp-controller #465: [tests] Fix flaky tests

pandafy, 2021-05-20:
I have observed that CI build for `Python==3.7 | django~=3.0.0` fails sometimes abruptly. 

One of such occurrence is https://github.com/openwisp/openwisp-controller/pull/421/checks?check_run_id=2622952481

Let's use this issue to report such flask tests and fix them all at once.

pandafy, 2021-06-01:
Build fail: https://github.com/openwisp/openwisp-controller/runs/2719061481

```
======================================================================
FAIL: test_ca_queryset (openwisp2.sample_pki.tests.TestAdmin) [ca2]
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/opt/hostedtoolcache/Python/3.7.10/x64/lib/python3.7/site-packages/openwisp_users/tests/utils.py", line 99, in _test_multitenant_admin
    msg_prefix='[operator not-contains]',
  File "/opt/hostedtoolcache/Python/3.7.10/x64/lib/python3.7/site-packages/django/test/testcases.py", line 465, in assertNotContains
    self.assertEqual(real_count, 0, msg_prefix + "Response should not contain %s" % text_repr)
AssertionError: 1 != 0 : [operator not-contains]: Response should not contain 'ca2'
```

nemesisdesign, 2021-07-23:
This build failed randomly but doesn't look like the same issue: https://github.com/openwisp/openwisp-controller/runs/3138770831

PR #694 (2022-09-16): [fix] Fixed flaky test_ca_queryset #465
The test was checking for a string that can be present in UUID of objects. Hence, it failed unexpectedly at times.

Fixes #465
commit 61ab5c0e (2022-09-16): [fix] Fixed flaky test_ca_queryset #465

The test was checking for a string that can be present in UUID of
objects. Hence, it failed unexpectedly at times.

Fixes #465
closed 2022-09-16 by nemesisdesign