openwisp/openwisp-controller #466: [bug] Validation failure on device page adds all templates to the device

pandafy, 2021-05-21:
Ensure that there are two different organizations and each organization has set of normal, enabled by default and required templates.

**Steps to replicate:**

1. Go to `/admin/config/device/add/`
2. Fill up all the fields(select an organization, add a configuration) but leave MAC Address field empty. (This will make the validation to fail)
3. Click on `Save and continue editing`
4. Now fill the MAC Address field and click on `Save and continue editing`

**Expected result**
The device should get saved/added

**Actual result**
The bellow validation error is shown

> The following templates are owned by organizations which do not match the organization of this configuration: Demo Enabled, Demo Required

Here `Demo *`  are templates that belong to the organization which is not selected for the device

You will also see an error in the console

[image]

**Root Cause**: All templates get added from the server when validation fails for adding new device. You can check by printing value of `$('input[name="config-0-templates"]').val()` on console. 

I think we should address this with https://github.com/openwisp/openwisp-controller/issues/446

_Originally posted by @pandafy in https://github.com/openwisp/openwisp-controller/issues/457#issuecomment-840071352_

pandafy, 2022-10-28:
@nemesisdesign I can no longer replicate this issue. Can you please confirm this?

nemesisdesign, 2022-10-28:
I can't replicate it too.

closed 2022-10-28 by nemesisdesign