openwisp/openwisp-controller #469: [bug] Unsaved changes alert triggered unnecessarily when required templates are in use

nemesisdesign, 2021-05-22:
### How to replicate

Add a required template to the system.
Create a new device in the same organization of the template just created.
Open the device detail page, do not change anything, reload or click on anywhere to go to a different page.

### Expected behavior

Device page is left without alerts.

### Actual behavior

Unsaved changes alert is triggered.

### Demo

[image]

### We need tests

This problem has been happening too much lately.
In order to avoid keep banging our head on it, I think we need to start introducing selenium tests, this could be the first (failing) test to add.

pandafy, 2021-06-02:
I was not able to replicate the bug described in the issue description. But I found another issue:

The required templates are always shown at top, irrespective of the ordering set by user.

nemesisdesign, 2021-06-02:
> I was not able to replicate the bug described in the issue description. But I found another issue:

I will double check.

> The required templates are always shown at top, irrespective of the ordering set by user.

That's the intended behavior and documented: https://github.com/openwisp/openwisp-controller#required-templates.

pandafy, 2021-06-02:
> That's the intended behavior and documented: https://github.com/openwisp/openwisp-controller/tree/797b41e1048f9b8c2939298c2f506a55e811e46b#required-templates.

But if you add them from shell using `Config.templates.add('<required-template-id>'), then they'll appear at the end of the list. Since this is a thing which won't happen in real world, I don't think we should worry about it.

pandafy, 2021-06-02:
I just noticed in the GIF attached in the issue description, the required template is not at the top.

nemesisdesign, 2021-06-02:
> > That's the intended behavior and documented: https://github.com/openwisp/openwisp-controller/tree/797b41e1048f9b8c2939298c2f506a55e811e46b#required-templates.
> 
> But if you add them from shell using `Config.templates.add(''), then they'll appear at the end of the list. Since this is a thing which won't happen in real world, I don't think we should worry about it.

Correct, they'll be reordered again sooner or later when the user hits save again on the same device.

nemesisdesign, 2021-06-02:
> I just noticed in the GIF attached in the issue description, the required template is not at the top.

Yes, that's the issue. I'll need to try again on an empty system.

nemesisdesign, 2021-06-03:
The problem happens on devices that exist before a required template is added to a system, I opened an issue to deal with required/default templates that should be automatically assigned to devices: https://github.com/openwisp/openwisp-controller/issues/480.

A problem with the unsaved changes code is still present but opened a dedicated issue for it: https://github.com/openwisp/openwisp-controller/issues/481.

closed 2021-06-03 by nemesisdesign