openwisp/openwisp-controller #477: [bug] Unsaved changes alert triggered when a new required template is added

pandafy, 2021-06-02:
**How to replicate**

1. Create a device with **default** organization
2. Create a required template for **default** organization
3. Open the device detail page, do not change anything, reload or click on anywhere to go to a different page.

**Actual behavior**

Unsaved changes alert is triggered.

**Reason**

The JS adds the required template created and updates the `config-0-templates` field

Aryamanz29, 2022-02-18:
Let me try this

DragnEmperor, 2025-04-14:
Hi @pandafy , I believe this might be a duplicate of #469 and should already be resolved, as I wasn’t able to reproduce the issue. If it’s still occurring, could you please share more context or steps to help replicate it?

nemesifier, 2025-07-09:
Thanks @DragnEmperor @Dhanus3133.

closed 2025-07-09 by nemesifier