openwisp/openwisp-controller #480: [change] Auto add default/required templates and make consistent with credentials auto_add

nemesisdesign, 2021-06-03:
When a template is flagged as default/required (either a new template or an existing non default/required is flagged as default/required), right now existing devices which would qualify to have the new template don't get it assigned automatically.

This is inconsistent with credentials with `auto_add` which do get added automatically to existing devices.

We should improve this and make the two features consistent and more explicit.

We may do it in different ways:

- simply warn the user that by enabling these features automatically adds the templates/credentials to any qualifying device
- add a confirmation page in which they are able to choose (but I wonder if this is really necessary, I personally don't see this as very useful)

Sreejith2, 2023-01-17:
I would like to contribute in this i am new to opensource can you please give me instructions?

mips171, 2024-05-02:
Got bitten by this today. Yeah you have to go to each device and "save" the config again. Not intuitive, since the config is already marked as requried.

devkapilbansal, 2025-05-05:
Haven't checked the current behaviour myself but ideally, we should have consistent behaviour.

nemesifier, 2025-05-05:
> [@nemesifier](https://github.com/nemesifier) I looked into it and found the following observations: **Devices created via Admin**
> 
> * Default templates: Not assigned automatically to either existing or new devices.
> * Required templates: Not assigned automatically to either existing or new devices.
> * Access Credentials (`auto add` : True): Not assigned automatically to either existing or new devices.

This is correct.

> **Devices created via REST API**:
> 
> * Default templates: Assigned automatically to new devices but not to existing devices.

Correct.

> * Required templates: Assigned automatically to both existing and new devices.

AFAIK this is not correct, are you sure @DragnEmperor?

> * Access Credentials (`auto add` : True):  Assigned automatically to both existing and new devices.

AFAIK this is not correct, are you sure @DragnEmperor?

> Are we expecting consistent behavior across all cases?

Yes, we must make it consistent.

pandafy, 2025-05-06:
> * Access Credentials (`auto add` : True): Not assigned automatically to either existing or new devices.

@DragnEmperor was the celery worker running when you performed this test?

DragnEmperor, 2025-05-07:
> This is correct.
>
>
> Correct.
>
>
> AFAIK this is not correct, are you sure [@DragnEmperor](https://github.com/DragnEmperor)?
>
>
> AFAIK this is not correct, are you sure [...]

@nemesifier Yes, attaching a video for reference

https://github.com/user-attachments/assets/da53a8b9-d0ec-4d84-bdd6-085c19dcf921

> > * Access Credentials (`auto add` : True): Not assigned automatically to either existing or new devices.
> 
> [@DragnEmperor](https://github.com/DragnEmperor) was the celery worker running when you performed this test?

@pandafy Yes, devices added via admin have no config thus leading to auto_add of access credentials excluding them.

pandafy, 2025-05-09:
> [@pandafy](https://github.com/pandafy) Yes, devices added via admin have no config thus leading to auto_add of access credentials excluding them.

This is expected since the `config.backend` is used to determine the `update_strategy` of the DeviceConnection. 
https://github.com/openwisp/openwisp-controller/blob/35dbfe5bb18b843bb0061af8cc9faebbf25db96f/openwisp_controller/connection/base/models.py#L297-L321

commit d79d252f (2025-07-15): [change] Auto add default and required templates to existing devices #480

Closes #480
commit bbce39d1 (2025-07-15): [change] Auto add default and required templates to existing devices #480

Closes #480
commit f3f88399 (2025-09-02): [change] Auto add default and required templates to existing devices #480

Closes #480
commit 8c65c898 (2025-09-02): [change] Auto add default/required templates to existing devices #480

Templates marked as default or required are now automatically
assigned to all eligible devices in a background task.

Closes #480
closed 2025-09-02 by nemesifier