openwisp/openwisp-controller #487: [qa] Enable css and js linting checks

niteshsinha17, 2021-06-10:
CSS and Js linting checks are disabled by default from ``openwisp-qa-check``. Check this https://github.com/openwisp/openwisp-utils/#openwisp-qa-check and enable it for ``controller`` and remove changes made in https://github.com/openwisp/openwisp-controller/pull/376 related to linting checks. Update readme too.

Red-0111, 2021-07-28:
Hi, I would like to work on this issue, please tell me how to proceed :)

harsh-9in, 2021-09-20:
hi, @niteshsinha17 I want to work on this issue. Can I proceed ?

nemesisdesign, 2021-09-20:
> CSS and Js linting checks are disabled by default from `openwisp-qa-check`. Check this https://github.com/openwisp/openwisp-utils/#openwisp-qa-check and enable it for `controller` and remove changes made in #376 related to linting checks. Update readme too.

@niteshsinha17 can you please refresh my mind about what has to be done here? Do we need to just switch to the linting done by openwisp-utils and remove any custom linting done in this repo?

niteshsinha17, 2021-09-21:
@nemesisdesign Yes, you are right. @harsh-9in go ahead :smile:

ashwini-m-hub, 2022-01-08:
still open can I do it

PR #576 (2022-02-12): [qa] Enable css and js linting checks #487
Fixes #487
commit f989c96a (2022-02-12): [ci] QA: Enabled css and js linting checks #487

Closes #487
closed 2022-02-12 by nemesisdesign