openwisp/openwisp-controller #491: [feature] Add API endpoint to return device group info based on a common name

nemesisdesign, 2021-06-13:
Follow up to #203.

We need  an API endpoint that is called by captive portals like PfSense to determine the captive page defined in the group meta data of OpenWISP.
The API endpoint will return the device group infomation, including the meta data (which in these legacy use cases will contain the captive page URL), given two input parameters: organizaton slug and certificate common name.

We need to implement this feature to provide backward compatibility to installations using OpenWISP 1.

### REST API

- [ ] Add a special API endpoint which returns the group information based on two input parameters: `organization_slug` and `common_name`; in short, given an organization slug and common name, we should find the x509 certificate with that common name and organization, find which vpnclient is associated to and hence find the device and return its group, if anything fails we'll return 404 a specific device based on the common name of a certificate associated as VPN client to a device, eg:
  `/api/v1/controller/{organization_slug}/certificate/{common_name}/group/`, we need to filter by organization becasue the same `common_name` could be in different orgs.
- [ ] We need to cache the response of this API endpoint because it will be called  whenever a user connects to the public wifi service, we need to invalidate the cache whenever the group is saved (`post_save` signal on `DeviceGroup` or when the device group changed signal built in #203 is emitted.

### Other cleanup

- [ ] Upon closer inspection I found out that `Cert.common_name` and `Ca.common_name` are not unique per organization so in this patch we should also change these two models so that thefields `common_name` and `organization` are flagged as `unique_together` (there should be a new way to do this in django)
- [ ] Write a test which tries to create duplicates of  certificates and a CAs having organization=None and a specific common_name, I am not sure if passing `organization=None` would skip the unique together check and we have to double check this

nemesisdesign, 2021-06-13:
PS: the organization passed as input cannot be only one unfortunately, we'll have to change this a bit to allow passing multiple organization slugs, look in all of them and return the first occurrence

pandafy, 2021-07-08:
>  We need to cache the response of this API endpoint because it will be called whenever a user connects to the public wifi service, we need to invalidate the cache whenever the group is saved/deleted, the certificate is saved/deleted or when the device group changed signal built in [controller/feature] Add DeviceGroup model #203 is emitted, explained in plain English: we have to invalidate the cache either when the group info is changed or the group is deleted, the group to which a device is assigned is changed or when the certificate is changed (eg: renewed) or deleted.

Can we use a index table in database to cache this? 
If we use combination of organization slugs and common name for cache key, we won't know what key to use for invalidation when either of organization, certificate or device group changes.

commit 1aaaa0b5 (2021-07-08): [feature] Added API endpoint to return device group using certificate common name #491

Closes #491
commit 1e93e083 (2021-07-08): [feature] Added API endpoint to return device group using certificate common name #491

Closes #491
commit 67f78ea5 (2021-07-15): [feature] Added API endpoint to return device group using certificate common name #491

Closes #491
commit 5b30934c (2021-07-26): [feature] Added API endpoint to return device group using certificate common name #491

Closes #491
closed 2021-07-27 by nemesisdesign