openwisp/openwisp-controller #502: [bug] jQuery conflict after merging DeviceGroup

nemesisdesign, 2021-07-16:
After upgrading an instance to the latest master which includes the DeviceGroup PR #492, the charts of OpenWISP monitoring broke with the following errors:

[image]

It seems also the flat json widget is being affected.

@pandafy can you please check?

nemesisdesign, 2021-07-17:
I confirm I can replicate the issue of the charts not loading also in the development environment, by using the latest openwisp-controller and run the dev server of openwisp-monitoring.

If I run the dev server of OpenWISP Controller, the error of the flat json widget shows although the widget seems to work anyway.

```
Uncaught TypeError: django.jQuery is not a function
    at flat-json-widget.js:160
```

pandafy, 2021-07-20:
@nemesisdesign can this be closed?

commit b7d87da5 (2021-07-20): [deps] Increased min django-flat-json version to 0.1.3 #502

Fixes #502
closed 2021-07-20 by nemesisdesign