openwisp/openwisp-controller #520: [validation/chores] Improve validation of subnet division rules

nemesisdesign, 2021-08-12:
- improve validation of size field:
  - backend: make sure a validation error is raised if size is changed for an existing rule (it should be already this way but let's double check)
  - frontend: if possible, use JS on an existing rule to disable the size field
- improve validation of number of subnets and IP addresses
  - backend: make sure a validation error is raised if the amount of IP or the amount of subnets in an existing rule is decreased (it should be already this way but let's double check)
  - frontend: if possible, use JS on an existing rule to prevent decreasing the amount of IP or the amount of subnets

pandafy, 2021-08-24:
I think we can allow modifying the number of IPs in either direction (increasing or decreasing). IPs are always provisioned from a subnet does not affect other provisioned IPs. There is a check which does not allow entering a value bigger than the provisioned subnet can accommodate. Hence, we should not put a restriction on updating number of IPs. 

We should think more about allowing to the change number of subnets. The main reason behind not allowing to update number of subnets were: 
1. if number of subnets is descreased, it will create patches of unused subnets in the master subnet. The implementation if not smart enough to utilise those patches in future provisioning 
2. if number of subnets is increased, the new subnets will be provisioned from the available portion of the master subnet. Thus, all provisioned subnet will not be continuous again. Now I don't think this will be problem since subnets/IPs will be referenced by their variable name most of the time.

pandafy, 2021-08-24:
Now I remember why we didn't allow reducing the number of IPs. Because it can break configurations. If a user has used identifier of an IP and it get's deleted. Those configurations will break.

nemesisdesign, 2021-09-28:
Closed by https://github.com/openwisp/openwisp-controller/pull/533.

commit 4bfcbb58 (2021-08-24): [enhancement] Improved validation of subnet division rules #520

Validation:
- Size of provisioned subnets cannot be changed
- Number of provisioned subnets cannot be changed
- Number of provisioned IP can only be increased

UX:
- If a subnet is not shared, organization fields of related
  Subnet Divsion Rule will be hidden and automatically set
  to organization of the subnet.

Closes #520
commit 14a6791a (2021-08-25): [enhancement] Improved validation of subnet division rules #520

Validation:
- Size of provisioned subnets cannot be changed
- Number of provisioned subnets cannot be changed
- Number of provisioned IP can only be increased

UX:
- If a subnet is not shared, organization fields of related
  Subnet Divsion Rule will be hidden and automatically set
  to organization of the subnet.

Closes #520
commit 8633eb78 (2021-09-15): [enhancement] Improved validation of subnet division rules #520

Validation:
- Size of provisioned subnets cannot be changed
- Number of provisioned subnets cannot be changed
- Number of provisioned IP can only be increased

UX:
- If a subnet is not shared, organization fields of related
  Subnet Divsion Rule will be hidden and automatically set
  to organization of the subnet.

Closes #520
PR #533 (2021-09-17): [enhancement] Improved validation of subnet division rules #520
Validation:
- Size of provisioned subnets cannot be changed
- Number of provisioned subnets cannot be changed
- Number of provisioned IP can only be increased

UX:
- If a subnet is not shared, organization fields of related
  Subnet Divsion Rule will be hidden and automatically set
  to organization of the subnet.

Closes #520
closed 2021-09-28 by nemesisdesign