openwisp/openwisp-controller #528: Support variables also in integer attributes

sklemer1, 2021-08-19:
I can't find a way to use variables for integer fields like wifi channels.

Either it is non-conforming json or it doesn't fit the schema.

nemesisdesign, 2021-08-19:
It's not supported now unfortunatley. I have plans to solve this in the future versions but it will take time.

kellerman, 2022-10-26:
Also stumbled upon this problem. Auto channel assignment is not really working (channel value '0' makes the radios just to get disabled) and integer variables doesn't work.
Creating more templates for each channel work fine as a workaround, but is a bit messy. It would be nice to see a better way to set numeric values.

nemesisdesign, 2022-10-27:
Depends on https://github.com/openwisp/netjsonconfig/issues/223.

nemesisdesign, 2022-10-27:
> Also stumbled upon this problem. Auto channel assignment is not really working (channel value '0' makes the radios just to get disabled)

BTW: this should not happen, OpenWrt can handle automatic channel selection via hostpad very well since many versions.

kellerman, 2022-10-27:
> > Also stumbled upon this problem. Auto channel assignment is not really working (channel value '0' makes the radios just to get disabled)
> 
> BTW: this should not happen, OpenWrt can handle automatic channel selection via hostpad very well since many versions.

I did not have the same luck with some Mikrotik boards. I'll recheck in a moment.