openwisp/openwisp-controller #529: [bug] Floorplan: vague "invalid value" validation error

nemesisdesign, 2021-08-19:
I found a bug while testing https://github.com/openwisp/openwisp-controller/pull/494.

Here's how to replicate it:

- create an outdoor location
- create a device, assign the outdoor location created in the previous step, save
- now open the device again, change the outdoor location to be indoor, add a new floorplan, specify floor number, specify an image, set a point on the image (to specify indoor coordinates), then save

Expected result:

It should get saved properly, but a generic "Invalid value" error is given to the floorplan coordinates, which isn't even shown.

I guess it's related to the fact that the type is being switched from outdoor to indoor.

It would be good to have some basic browser testing for this feature but I am not sure whether we should do this here or in django-loci.

DragnEmperor, 2025-03-13:
Hi @nemesifier, i looked into this and i think it is due to 'type' of form not being synced with location type. We can do this in django-loci

nemesifier, 2025-03-13:
@DragnEmperor sure, go ahead!

DragnEmperor, 2025-03-14:
@nemesifier I have duplicated the issue in django-loci repo as per template and also opening a PR for the same. 
Issue: https://github.com/openwisp/django-loci/issues/156
PR: https://github.com/openwisp/django-loci/pull/157
Please let me know if i am missing something or if there's something in the approach

nemesifier, 2025-04-02:
I have verified that after merging https://github.com/openwisp/django-loci/pull/157, with the latest version of django-loci the bug is fixed, thank you very much @DragnEmperor 🙏

closed 2025-04-02 by nemesifier