openwisp/openwisp-controller #530: [bug] Now floorplan exists for this location

nemesisdesign, 2021-08-19:
- Create an indoor location
- Create a floorplan related to this location
- Assign the location to a device
- Switch the location from indoor to outdoor, save
- Open the device page, go to map, then change its type again from outdoor to indoor, then try to load existing floorplans

Expected result:

The floorplan created at step 2 is shown.

Actual result: no floorplans exist for this location.

Would be good to have a browser test for this.

dee077, 2025-01-23:
On creating a indoor floorplan related to a location and then changing it to outdoor it shows error
`Please remove the associated floorplans first and save; then you can switch to type "indoor"` (error message is also wrong it should be  switch to type "outdoor")
but when when we change it via device you are allowed to change it and save as well, isn't it should show the same error in device as well?

Because it is causing the issue, when the location is set to outdoor via device  it is not supposed to have floorplan (in django-loci we are just returning when location is outdoor in `indoorForm funtion in loci.js`) and that's why is shows error `no floorplans exist for this location`.

Solution: we can add the js logic when we change location from indoor which has a floorplan to outdoor shows the same error message as above, which is already been written to  `AbstractObjectLocationForm` in `django-loci` and write selenium test for it here

Please let me know your thoughts on this!

nemesifier, 2025-01-31:
@dee077 yes, I think the solution you are proposing is good.

DragnEmperor, 2025-04-13:
@nemesifier  since we are now deleting floorplans on location type conversion, we should ideally not encounter this bug.

devkapilbansal, 2025-05-05:
Agree, this bug should have been fixed by https://github.com/openwisp/django-loci/pull/157