openwisp/openwisp-controller #573: [bug] Unsubscribe button in device page is slightly higher than other buttons

nemesisdesign, 2021-11-24:
[image]

Can't be seen easily but if we focus our eyes on the lower part of the button we'll notice it's slightly higher than the other buttons, let's fix this.

adii21-Ux, 2021-11-25:
Can I work on this?

codesankalp, 2021-11-25:
> Can I work on this?

Yes, go ahead :+1:

subhamgcon, 2021-12-20:
Hey @codesankalp Is this issue still open or resolved?

codesankalp, 2021-12-20:
> Hey @codesankalp Is this issue still open or resolved?

It's still open.

rakite28, 2022-01-06:
hey i am new to this i want to help resolve this can u give me some idea as to how i can move forward in this i have cloned the repo already and know a little bit about js

codesankalp, 2022-01-06:
> hey i am new to this i want to help resolve this can u give me some idea as to how i can move forward in this i have cloned the repo already and know a little bit about js

@rakite28 explore and setup the repo first. when you will get familiar with this, you will be able to solve this.

rakite28, 2022-01-06:
👍👍thanks I will try and familiarise myself with it

ashwini-m-hub, 2022-01-08:
its still open can I do

OsmiumOP, 2022-01-18:
Can you assign this to me?

codesankalp, 2022-01-18:
@OsmiumOP You can start working without assigning if @ashwini-m-hub is not working on this. More issues are present here: https://github.com/orgs/openwisp/projects/4

pandafy, 2022-01-25:
[image]
[image]

Heights of unsubscribe and other buttons is different. It might be due to the notification bell icon.

sahilsaini110, 2022-02-17:
is this issue still open

pandafy, 2022-02-17:
@sahilsaini110 there's already a PR open for this.

pandafy, 2022-02-17:
This has been fixed in https://github.com/openwisp/openwisp-notifications/pull/220