openwisp/openwisp-controller #577: [bug] Restoring deleted device fails in prod

nemesisdesign, 2021-12-07:
Modules enabled:

- fw upgrader
- topology
- monitoring

After deleting a device, the restore fails because of a validation error which is not shown in the UI.

okraits, 2021-12-07:
We once experienced something like that as well. It could be that the cause is that some related object could not be set back to the chosen state. 
You have to be careful when using django-reversion. Better do atomic versions where each change is versioned in a separate revision.

pandafy, 2021-12-22:
**Steps to replicate**

1. Create an Access Credential and flag it for auto adding. 
2. Create a device and ensure that DeviceConnection object is created for this device.
3. Delete the device.

**Actual result**
The error mentioned in the original comment appears.

commit db85f855 (2021-12-23): [fix] Fixed error on restoring device #577

Closes #577
commit 6c5ee0bd (2021-12-27): [fix] Fixed error on restoring device #577

Closes #577
commit 92603efd (2021-12-29): [fix] Fixed error on restoring device #577

Closes #577
PR #581 (2022-01-05): [fix] Fixed error on restoring device #577
Closes #577
closed 2022-01-05 by nemesisdesign