openwisp/openwisp-controller #592: [bug] Encryption label missing in configuration editor

pandafy, 2022-01-21:
**Steps to replicate**

1. Create a device and  add a config to it. 
2. Add a "Wireless Interface" using the configuration editor. 

**Actual Result**

Encryption and Access Point labels are missing
[image]
[image]

**Expected Result**
These labels should be visible

pandafy, 2022-01-21:
I suspect these changes are causing this issue

nemesisdesign, 2022-01-21:
https://github.com/openwisp/openwisp-controller/commit/56ccd8c61d5c7f35b90c9b9d8fc754e78ca59ca8#r64349961

commit d25bf51d (2022-03-12): [fix:ui] Fixed config switcher positioning #592

Fixes #592
closed 2022-03-12 by nemesisdesign