openwisp/openwisp-controller #597: [bug] VpnClient deletion does not respect `auto_cert` flag

pandafy, 2022-01-28:
Removing a VPN Client template from a device will delete the related VpnClient object. This in-turn deletes related `Cert` objects due to `on_delete=models.CASCADE`. 

Instead of using database cascading, we can add a `post_delete` handler that only deletes related certificates if `auto_cert` field is set to `True`.

yashikajotwani12, 2022-01-31:
I would like to work on this.

pandafy, 2022-01-31:
Hi @yashikajotwani12! This is an advance issue that will require some discussion before implementing a solution. Consider taking up another issue.

Thanks!

yashikajotwani12, 2022-01-31:
> 

Okay! sure. I will look for another issue. Thanks!

yashikajotwani12, 2022-01-31:
> Hi @yashikajotwani12! This is an advance issue that will require some discussion before implementing a solution. Consider taking up another issue.
> 
> Thanks!

Okay, sure! I will look for another issue. Thanks

commit ce980659 (2022-12-30): [fix] Fixed VpnClient deletion does not respect auto_cert flag #597

Fixes #597
PR #732 (2023-01-11): [fix] Fixed `VpnClient` deletion does not respect `auto_cert` flag #597
Fixes #597
commit ce8d27c4 (2023-01-11): [fix] Fixed `VpnClient` deletion does not respect `auto_cert` flag #597

Fixes #597
closed 2023-01-11 by nemesisdesign