openwisp/openwisp-controller #611: [bug] Deleting a VPN Client certificate raises 500 error

pandafy, 2022-02-23:
**Steps to replicate**

1. Create a VPN server with _OpenVPN_ backend 
2. Create a template for the VPN server created in step 1 
3. Create a device and apply the template created in step 2 to this device 
4. Go to `/admin/pki/cert/` and delete the certificate related to the device

**Expected Outcome**
The certificate is deleted and along with it VpnClient object is also deleted. 

**Actual Outcome**
Server 500 Error

Possible fix: 
Wrap the following code in a try except block. 

https://github.com/openwisp/openwisp-controller/blob/758e398a60a904ed688f345b03924289e934ccd6/openwisp_controller/config/base/vpn.py#L660-L661

We need a dedicated test for this regression.

Subhraneel77, 2022-02-27:
Can I Give it a Try?

pandafy, 2022-03-01:
Sure! Go ahead @Subhraneel77

Aryamanz29, 2022-05-07:
I'd like to work on this.

nemesisdesign, 2022-05-07:
@Aryamanz29 :+1: :+1:

commit 7404c1a8 (2022-05-09): [fix] Fixed deletion of VPN Client certificate #611

- Deletion of VPN Client certificate raises 'ObjectDoesNotExist' exception.
- Wrapped post_delete signal in a try-except block.

Closes #611
closed 2022-05-09 by nemesisdesign