openwisp/openwisp-controller #631: [feature] Add group templates

nemesisdesign, 2022-04-19:
## Group templates

The possibility to select templates which would automatically be assigned to devices when devices are moved into the group will be added.

The system will make sure to reuse the relevant_templates.js script and any related logic to show only templates which are relevant to the currently selected organization.

## Exclusion of default/required templates

The system will not show templates which are flagged as default or required because that would be redundant since those templates would be assigned to new devices anyway.

**The help text of the field will explain this properly**.

Here’s a preview of how the Group admin page would look like:

[image]

## Automatic removal of group templates on group removal/change

When the group of a device changes the system will automatically remove the templates of the group from the device configuration.

We shall leverage the `device_group_changed` which was already introduced when the device group feature was first implemented.

## Docs

We shall add a mention of this feature in the usage section of the README (group subsection) plus a section with a few screenshots that shows how to use it.

codesankalp, 2022-05-16:
> We shall leverage the [device_group_changed](https://github.com/openwisp/openwisp-controller#device-group-changed) which was already introduced when the device group feature was first implemented, but since that signal doesn't fire when the device is created, we shall also use `Device.post_save` and check whether the device has just been created and if it has a group defined, if these 2 conditions are met we can trigger the group template logic on `post_save` too.

I checked that `device_group_changed` gets fired on creation of device also if device group is selected while creating the device. So, I think we don't need to implement post_save signal for device model.

nemesisdesign, 2022-05-16:
> I checked that `device_group_changed` gets fired on creation of device also if device group is selected while creating the device. So, I think we don't need to implement post_save signal for device [...]

Thanks for reporting this, I created https://github.com/openwisp/openwisp-controller/issues/649 to clarify whether this is a bug that needs to be fixed or a documentation issue.

pandafy, 2022-05-17:
> I checked that `device_group_changed` gets fired on creation of device also if device group is selected while creating the device. So, I think we don't need to implement post_save signal for device model.

@codesankalp can you share steps for replicating this issue?

PR #645 (2022-06-27): [feature] Add group templates #631 
Closes #631
commit 126ab584 (2022-06-27): [feature] Add group templates #631

Implements and closes #631
closed 2022-06-27 by nemesisdesign