openwisp/openwisp-controller #671: Daemon service doesn’t autostart after upgrading firmware via OpenWISP

sergorl, 2022-07-06:
I made custom package with daemon service:

```
#!/bin/sh /etc/rc.common
 
exec >> /root/log 2>&1
echo "STARTED non-lora-aggregator"
set -x

USE_PROCD=1
 
START=99
STOP=99

PROG="/usr/bin/non-lora-aggregator"
 
start_service() {
    procd_open_instance
    procd_set_param command $PROG
    procd_set_param stdout 1
    procd_set_param stderr 1
    procd_close_instance
}
```
and included it in final firmware.

After command `sysupgrade -v my-firmware.bin` my daemon service `non-lora-aggregator` autostarts and works OK.

But when I upgrade my OpenWRT device with the same my-firmware.bin via [OpenWISP](https://openwisp.org/) my daemon service `non-lora-aggregator` doesn't start. By the way, after sending command reboot to my device via the same [OpenWISP-controller](https://openwisp.io/docs/user/firmware-upgrades.html) my daemon starts after rebooting the device.

My expectations: all daemon services should autostart despite how I do firmware upgrading via `sysupgrade -v my-firmware.bin` or [OpenWISP-controller](https://openwisp.io/docs/user/firmware-upgrades.html)

Why it happens? Please, help me to fix it.

okraits, 2022-07-07:
This issue should rather be in https://github.com/openwisp/openwisp-firmware-upgrader.

I doubt that this has anything to do with OpenWISP, though. `openwisp-firmware-upgrader` also uses `sysupgrade -v <file>` to flash the given firmware:

https://github.com/openwisp/openwisp-firmware-upgrader/blob/master/openwisp_firmware_upgrader/upgraders/openwrt.py#L28

Did you check the syslog after upgrading via `openwisp-firmware-upgrader`?

sergorl, 2022-07-07:
@okraits , I republished it here https://github.com/openwisp/openwisp-firmware-upgrader/issues/201
So I close it.

closed 2022-07-07 by sergorl