openwisp/openwisp-controller #673: rest api to send command seems doesn't work

sergorl, 2022-07-08:
My `curl` command looks like:

```bash
curl -X POST "https://openwisp.internal.chirpwireless.io/api/v1/controller/device/8cb65d82-dfd8-4a36-94ca-206aec315494/command/" -H "accept: application/json" -H  "Authorization: Bearer "some token"" -H  "X-CSRFToken: lhYJJWLgB53saQaOZXWMSxYSe38pAghCHFivFYW9TuOpBGFi9ubE2s1OIfplDoGh" -d "{  \"input\": \"echo \"Hello from OpeWISP\" >> /data/data.txt\",  \"connection\": \"ca39961c-d3ca-4e0a-892f-637d29468007\",  \"type\": \"custom\"}"

```

But I got this: `{"input":["This field is required."],"type":["This field is required."]}`

what am I doing wrong?

nemesisdesign, 2022-07-08:
@sergorl please check the log lines generated right after the request which generates 500 error in `/opt/openwisp2/log/openwisp2.log` and report here.

sergorl, 2022-07-11:
`/opt/openwisp2/log/openwisp2.log`:

```bash
File "/opt/openwisp2/env/lib/python3.8/site-packages/django/db/models/base.py", line 1405, in full_clean
    raise ValidationError(errors)
django.core.exceptions.ValidationError: {'input': ['Enter valid JSON.', "'mkdir /test_openwisp' is not of type 'object'"]}
```

I fixed it by this `-d "{  \"input\": {\"command\": \"mkdir /test_openwisp\"},  \"connection\": \"ca39961c-d3ca-4e0a-892f-637d29468007\",  \"type\": \"custom\"}"` - I found it via chrome developer tools: 
```bash
curl -X POST "https://openwisp.io/api/v1/controller/device/8cb65d82-dfd8-4a36-94ca-206aec315494/command/" -H "Content-Type: application/json" -H  "Authorization: Bearer "some token"" -H  "X-CSRFToken: lhYJJWLgB53saQaOZXWMSxYSe38pAghCHFivFYW9TuOpBGFi9ubE2s1OIfplDoGh" -d "{  \"input\": {\"command\": \"mkdir /test_openwisp\"},  \"connection\": \"ca39961c-d3ca-4e0a-892f-637d29468007\",  \"type\": \"custom\"}"
```

P.S.: It seems your swagger `/api/v1/docs/` lies:
it generates curl with `-H "accept: application/json"` but it doesn't work. `-H "Content-Type: application/json"` works.

closed 2023-05-29 by nemesisdesign