openwisp/openwisp-controller #680: [problem] On SSH connection errors failure reason indicates failure to connect to public IP

nemesisdesign, 2022-07-15:
When SSH connections are attempted, the management IP is tried first, the `last_ip` is tried as last.
If both failed, the error reason is the failure to the last IP.

It's normal to think that the system is trying to connecto the `last_ip`, but that is not the case.
The error message is misleading.

Maybe a better approach could be to show both failures.

Tryinig the ` last_ip` is needed for situations in which OpenWISP can reach the devices on `last_ip` too.

However, another approach could be to decide that OpenWISP shall try reaching devices only using the `management_ip`.
The monitoring module has a setting for this. In this case we should rework a few things here too.

nemesisdesign, 2023-05-31:
We must verify wether this issue is still relevant.

nemesifier, 2023-06-29:
The introduction of OPENWISP_CONTROLLER_MANAGEMENT_IP_ONLY set to True has fixed this issue, if that setting is set to False we assume last_ip is reachable by OpenWISP so if it fails it's ok to show its failure.

closed 2023-06-29 by nemesifier