openwisp/openwisp-controller #682: [bug] 500 internal server error for NoReverse match

nemesisdesign, 2022-07-19:
Related to #681 (happens while testing that bug).

Try opening `http://localhost:8000/admin/config/device/de8fa775-1134-47b6-adc5-2da3d0626c72/history/1564/undefinedadmin/img/icon-deletelink.svg`.

Expected result: 404 not found
Actual result: 500 internal server error

```
NoReverseMatch: Reverse for 'config_template_download' with arguments '('de8fa775-1134-47b6-adc5-2da3d0626c72/history/1564/undefinedadmin/img/icon-deletelink.svg',)' not found. 1 pattern(s) tried: ['admin/config/template/download/(?P<pk>[^/]+)/$']
  File "django/core/handlers/exception.py", line 55, in inner
    response = get_response(request)
  File "django/core/handlers/base.py", line 197, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
  File "django/contrib/admin/options.py", line 683, in wrapper
    return self.admin_site.admin_view(view)(*args, **kwargs)
  File "django/utils/decorators.py", line 133, in _wrapped_view
    response = view_func(request, *args, **kwargs)
  File "django/views/decorators/cache.py", line 62, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)
  File "django/contrib/admin/sites.py", line 242, in inner
    return view(request, *args, **kwargs)
  File "openwisp_controller/config/admin.py", line 133, in change_view
    extra_context = self.get_extra_context(object_id)
  File "openwisp_controller/config/admin.py", line 110, in get_extra_context
    ctx['download_url'] = reverse('{0}_download'.format(prefix), args=[pk])
  File "django/urls/base.py", line 88, in reverse
    return resolver._reverse_with_prefix(view, prefix, *args, **kwargs)
  File "django/urls/resolvers.py", line 802, in _reverse_with_prefix
    raise NoReverseMatch(msg)
```

It looks like the URL pattern for this route is not strict enough.

The same happens with Template and VPN.

I think this is probably happening for all the url patterns, we should change the pattern and only allow letters, numbers and dashes, nothing more. The `re_path` must be kept for backward compatibility reasons (devices send the UUID without dashes and that is not recognized by Django's normal `path` functions).

praptisharma28, 2025-01-31:
[image]

Hey @pandafy , @nemesifier do we want this default message to be shown, or should we add a custom message instead?

nemesifier, 2025-01-31:
@praptisharma28 returning 404 is fine, the URL shouldn't be recognized in the first place, I suspect there's something wrong with some definition in the `urls.py` file.