openwisp/openwisp-controller #700: API /user/organization missing  "Configuration Management Setttings"

clutch2sft, 2022-10-11:
Using this API endpoint in any of its variants does not allow setting or getting of the "Shared secret."  

I don't know enough about the internals to say much about this problem.

Expected: to be able to set the shared secret when using the endpoint for ORG creation, set the shared secret when using the PUT or PATCH, and retrieve the shared secret when using GET on the /user/organization endpoint.  

-OR- provide another endpoint as necessary to allow the setting and getting of the ORG shared secret depending on internals and what is needed to affect this outcome.

Current - no access to ORG shared secret from any API endpoint.

AyushKhatri-Dev, 2026-09-27:
I’d like to work on this. I read #1208 — it added a full /controller/organization/ endpoint, which means Organization is exposed in two places.

I think the endpoint should stay under controller since OrganizationConfigSettings lives there. Instead of duplicating the whole Organization endpoint, a sub-endpoint just for the config settings feels cleaner. What do you think?

AyushKhatri-Dev, 2026-09-28:
@openwisp-companion assign