openwisp/openwisp-controller #703: [bug] Group templates feature causes VPN client recreated every time config is changed

nemesisdesign, 2022-11-02:
Somewhat similar to https://github.com/openwisp/openwisp-controller/issues/399 but triggered by a different cause.

**How to replicate:**

- make sure to have at least a group
- make sure to have a VPN server with openvpn and auto cert and a VPN template
- create a device assigned to the group, create a config for it and add the VPN template to it, save, reload the change page
- ensure the cert is created, open the cert from the cert management page
- now change a detail of the config (eg: variables)

**Expected result:**

- cert is not destroyed and recreated
- `openwisp_controller.config.tasks.invalidate_devicegroup_cache_delete` is not triggered
- `openwisp_controller.config.tasks.invalidate_devicegroup_cache_change` is not triggered

**Actual result:**

- cert is not destroyed and recreated
- `openwisp_controller.config.tasks.invalidate_devicegroup_cache_delete` is not triggered
- `openwisp_controller.config.tasks.invalidate_devicegroup_cache_change` is not triggered

I spent a considerable amount of time debugging this and concluded the culprit is this method:

https://github.com/openwisp/openwisp-controller/blob/e815119c8a4a62ef2db34a95a488e676572720ab/openwisp_controller/config/base/config.py#L626-L642

Removing and adding the templates again is not the best solution, we must look into an alternative.
@codesankalp do you remember why do we have to remove and add again all the templates? Was it just because at the time this seemed the simplest solution?

codesankalp, 2022-11-04:
> **Expected result:**
>
>     * cert is not destroyed and recreated
>
>     * `openwisp_controller.config.tasks.invalidate_devicegroup_cache_delete` is not triggered
>
>     * [...]

Why the expected and actual result is same?

>Removing and adding the templates again is not the best solution, we must look into an alternative.
@codesankalp do you remember why do we have to remove and add again all the templates? Was it just because at the time this seemed the simplest solution?

I don't remember the actual reason of doing this, but I think it is due to call respective `m2m_changed` signals.

Also, I think calling this function in save is causing this bug:
https://github.com/openwisp/openwisp-controller/blob/e815119c8a4a62ef2db34a95a488e676572720ab/openwisp_controller/config/admin.py#L365-L378

nemesisdesign, 2022-11-05:
> Why the expected and actual result is same?

@codesankalp a typo sorry, I amended, I meant to write that the actual result is that certs are destroyed and recreated.

It's these 2 lines:

```python
self.templates.remove(*old_templates) 
self.templates.add(*templates)
```

pandafy, 2022-11-25:
Hey @codesankalp, are you still working on this issue?

codesankalp, 2022-11-27:
> Hey @codesankalp, are you still working on this issue?

No @pandafy, If it's urgent, You can go ahead.
I am planning to do this after next week.

commit 3c8cb000 (2022-11-28): [fix] Fixed group templates re-creating VPN clients #703

Fixes #703
commit ddac15a4 (2022-11-28): [fix] Fixed group templates re-creating VPN clients #703

Fixes #703
PR #723 (2022-11-28): [fix] Fixed group templates re-creating VPN clients #703
Fixes #703
commit 2caff648 (2022-11-28): [fix] Fixed group templates re-creating VPN clients #703

Fixes #703
closed 2022-11-28 by nemesisdesign