openwisp/openwisp-controller #719: [feature] support multi-factor authentication

mips171, 2022-11-21:
This could be useful especially in multi-tenant and commercial installations to help protect customer data. Ideally this should support authenticator apps. I have seen some Django packages such as [this one](https://github.com/jazzband/django-two-factor-auth) that may be able to deliver a turnkey solution. 

[Documentation](https://django-two-factor-auth.readthedocs.io/en/1.14.0/configuration.html#general-settings)

nemesisdesign, 2022-11-22:
There's high chances that this feature does not need any new code to be added to OpenWISP in order to work but this needs to be confirmed.

aoutifrak, 2025-03-29:
is the authentication needs to be in the login?