openwisp/openwisp-controller #731: [bug] Recover deleted device button disappeared

nemesisdesign, 2022-12-13:
In the template list page the button is showing:

[image]

In the device list page the button disappeared, probably overridden by the import/export feature?

Aryamanz29, 2022-12-19:
Hey @nemesisdesign, In the device list page, I'm able to see those buttons. See below:

[image]

nemesisdesign, 2022-12-19:
You're right @Aryamanz29, the problem appears in the dev env of OpenWISP Monitoring, must be something that has to do with the order of INSTALLED_APPS, we'll have to create an issue for ansible-openwisp2 and docker-openwisp too because it shows up in deployed instances as well.