openwisp/openwisp-controller #738: [feature] Convert group metadata to variables

nemesisdesign, 2023-01-23:
The group metadata which is now retrievable via API, could be used as variables in the configuration of devices belonging to the group, which now is not the case.

I think we could keep the internal naming "meta data" for backward compatibility but could probably add a help text in the UI to explain that it can be used either for metadata and config variables.

The meta data field then shall be included in the device config context, we shall add a test and update the README to mention this feature.

theVJagrawal, 2023-01-28:
hey @nemesisdesign I want to work on this issue so can you please elaborate it more

pandafy, 2023-06-26:
I and @nemesifier discussed the following changes to be implemented:

- Rename `meta_data` field to `device_group`
- In the `DeviceGroupDetailView`  return both `meta_data` and `context` fields. The `meta_data` field will mirror the output of the `context` field. The `meta_data` field would be flagged as deprecated.  
- Change the widget of DeviceGroup.context field to use FlatJsonWidget. The field should be called "Configuration Variables" in the UI. This will make it consistent with implementation of configuration variables in the device admin.
- On the device change page, the `System Defined Variables` shall also contain the configuration variable in the selected device group. 
- We shall update the `get_template_default_values` view to also accept `group` URL parameter. If a template and device group defines the same configuration variable, then device group takes precedence. (Only override the conflicting keys)

nemesifier, 2023-06-26:
> * Rename `meta_data` field to `device_group`

I think we said `context`.

> * In the `DeviceGroupDetailView`  return both `meta_data` and `context` fields. The `meta_data` field will mirror the output of the `context` field. The `meta_data` field would be flagged as deprecated.

Yes.

> * Change the widget of DeviceGroup.context field to use FlatJsonWidget. The field should be called "Configuration Variables" in the UI. This will make it consistent with implementation of configuration variables in the device admin.

Yes, we need to make sure any non string value is converted to string.

> * On the device change page, the `System Defined Variables` shall also contain the configuration variable in the selected device group.

Yes.

> * We shall update the `get_template_default_values` view to also accept `group` URL parameter. If a template and device group defines the same configuration variable, then device group takes precedence. (Only override the conflicting keys)

Yes.

pandafy, 2023-06-27:
> * Change the widget of DeviceGroup.context field to use FlatJsonWidget. The field should be called "Configuration Variables" in the UI. This will make it consistent with implementation of configuration variables in the device admin.

@nemesifier when discussing about this change yesterday, we didn't take [OPENWISP_CONTROLLER_DEVICE_GROUP_SCHEMA](https://github.com/openwisp/openwisp-controller#openwisp_controller_device_group_schema) into account. If we change the widget to `FlatJsonWidget`, then we won't be able to provide a form. 

One option is to drop that setting altogether for simplicity. 

The other option which I can think right now is as follows (thinking out loud): 

We provide a default schema for `OPENWISP_CONTROLLER_DEVICE_GROUP_SCHEMA`: 
```json

{
  "type": "object",
  "properties": {
    "configuration_variables": {
      "type": "object",
      "title": "Configuration Variables",
      "additionalProperties": {
        "type": "string"
      }
    },
    "meta_data": {
        "type": "object",
        "title": "Metadata",
    }
  },
}
```

This would generate the following UI:

[image]

This breaks the consistency with other admin pages (device and template) - **A major drawback**

Instead of going ahead of the second option, I think it would be better to add a new field `context` to the DeviceGroup. We separate the usage of `context` and `meta_data` field 

- `context`: key-value pairs that should be accessible by the device configuration 
- `meta_data`: some additional data (data about data), for the device which is not directly used by the device 

But then, `meta_data` would become absurd. A data about device which is not accessed by the device at all ???

commit 1227355d (2023-04-27): [feature] Convert group metadata to variables #738

Closes #738
commit 9cb3c7c4 (2023-07-10): [feature] Added configuration variables to DeviceGroup #738

Closes #738
commit e53b82f9 (2023-07-13): [feature] Added configuration variables to DeviceGroup #738

Closes #738
PR #779 (2023-07-14): [feature] Added configuration variables to DeviceGroup #738
Closes #738
closed 2023-07-14 by nemesifier