openwisp/openwisp-controller #744: [bug] Defining OpenVPN Server complains about validation and does not allow saving

nemesisdesign, 2023-02-21:
Steps:

- create a new template
- add openvpn conf
- select server routed
- prepare a sample config
- try saving

Expected result: can save.

Actual result: validation error doesn't allow saving.

anshhh713, 2026-01-01:
I can reproduce this issue locally.

I’d like to take a stab at improving the UX around required configuration by making it clearer in the UI and preventing submission until configuration is present.

I’ll update here if I run into any issues.

anshhh713, 2026-01-02:
I tested this with a completely fresh setup:

- Forked the repository on a new account

- Fresh clone

- docker compose build

- docker compose up

- No existing data (new DB / volumes)

Following the same steps described in this issue, I was not able to reproduce the validation error — saving the template succeeded.

However, in a previous local setup (older fork / existing data), performing similar steps resulted in the generic validation error (“Please correct the error below”), even when following the same flow.

This suggests the issue may be state-dependent (e.g. related to existing OpenVPN configs, templates, or prior data), rather than strictly tied to the creation order.

I no longer have screenshots from the previous setup, but I can try to reproduce it again by reintroducing existing data if that would help.

**Additional observation**:
In the previous setup, saving the template initially worked without any issues.
Later, without changing the steps, saving started failing with the same generic validation error.
This occurred irrespective of creation order or server mode (bridged / routed / manual), and once the error started occurring, it persisted for subsequent attempts in that environment.

rohanesor, 2026-10-20:
I spent some time investigating this. Summary: a correct "Server (routed)" config **does** validate — which is why @anshhh713 couldn't reproduce it from a clean setup. The actual defect is in how the `oneOf` in the openvpn schema reports failures.

### Reproduction

Model layer is not at fault. `Template` (generic/OpenWrt) and `Vpn` (`vpn_backends.OpenVpn`) both pass `full_clean()` with routed, bridged and manual server configs.

The openvpn item schema is:

```
items.oneOf = [client, server_manual, server_bridged, server_routed]
items.additionalProperties = true
```

Validating a config against each branch individually:

| config | client | server_manual | server_bridged | server_routed | matched |
|---|---|---|---|---|---|
| correct routed config | ✗ | ✗ | ✗ | ✓ | 1 → valid |
| routed + leftover `server_bridge` key | ✗ | ✗ | ✓ | ✓ | **2 → fails** |
| `server: "10.8.0.0"` (under `minLength: 15`) | ✗ | ✗ | ✗ | ✗ | **0 → fails** |

**Failure mode 1 — a legitimate config is rejected.** Because `additionalProperties` is `true`, switching the editor from *Server (bridged)* to *Server (routed)* leaves the old `server_bridge` key in the JSON. The config then satisfies two branches, and `oneOf` requires exactly one. This is reproducible through the UI but not by pasting clean JSON, which I think explains the conflicting reports here.

**Failure mode 2 — the real error is discarded.** A small mistake such as a too-short `server` value produces a precise error inside the `oneOf` context, but the user only sees the whole config dumped followed by *"is not valid under any of the given schemas"*, with no indication of which property is wrong.

### On #1215

For what it's worth, this is consistent with @nemesifier's review there: attaching the same message to the `config` field doesn't make it readable, and it doesn't address failure mode 1 at all.

### Possible directions

For mode 2, the failing `oneOf` still carries its per-branch errors in `jsonschema` (`err.context`). Selecting the sub-error from the branch that got furthest recovers an actionable message — verified locally:

```
before : {'name': 'srv', 'dev': 'tun0', ...} is not valid under any of the given schemas
after  : openvpn.0.server: '10.8.0.0' is too short
```

(`jsonschema.exceptions.best_match` is not suitable here — it selects the `client` branch and reports `'server' is not one of ['p2p']`, which is misleading.)

For mode 1, the options I can see are making the server branches mutually exclusive in the schema, or having the editor drop keys belonging to the previously selected branch.

### Question before I write any code

1. Is the goal a comprehensible error message, accepting the ambiguous-but-legitimate config, or both?
2. Should this be fixed in `clean_netjsonconfig_backend()` here, or upstream in netjsonconfig (where the schema and `validate()` live)?

Happy to implement whichever direction you prefer — I'd rather confirm the scope first than repeat the approach that was rejected in #1215.