openwisp/openwisp-controller #769: [change] Make the host field of the VPN model configurable for ZeroTier Controllers

Aryamanz29, 2023-05-22:
We should make the `host` field of the VPN model configurable to support both [ZeroTier Central Controllers](https://discuss.zerotier.com/) &  [ZeroTier Self-Hosted controllers](https://docs.zerotier.com/self-hosting/introduction).

Aryamanz29, 2023-05-23:
@pandafy https://github.com/openwisp/openwisp-controller/issues/605#issuecomment-1042057468
> `https://my.zerotier.com/` in above points should be made configurable using the `Host` field of VPN model to support self-hosted ZeroTier controllers.

pandafy, 2023-10-11:
In this iteration, we have added support for the self-hosted version of ZeroTier APIs. The host field is already editable in the VPN object. 

This issue can be closed without requiring any further work.

closed 2023-10-11 by pandafy