openwisp/openwisp-controller #777: [feature] New sortable fields in device list

tricksol, 2023-06-14:
It would be very nice to have "Config Status" and "IP address" as sortable fields.

DragnEmperor, 2025-03-14:
For “Config Status”, since it’s a choice field, we can assign a key to each choice and sort based on those keys.
For “IP address”, we can sort numerically rather than lexicographically to maintain the correct order (e.g., 192.168.1.100 should come after 192.168.1.2). 
Could you please confirm if this approach aligns with your requirements?

tricksol, 2025-03-14:
Yes that would work, I'm not sure if this has been done yet but if the ip address could be a click able field to open a new browser tab would be helpful for my case as well.

DragnEmperor, 2025-03-15:
Okay Cool!