openwisp/openwisp-controller #788: [feature] Allow defining Wireguard peers in the server to reserve IP addresses

nemesifier, 2023-07-07:
Right now to allow non OpenWISP peers to connect to a Wireguard server managed by OpenWISP, the user needs to create fake devices which act as placeholders and reserve the IP addresses needed by these clients.

We could make this easier in the following way:

- when peers are defined in the server configuration automatically create the IP address entries in IPAM
- we have to find a way to remove the IPAM entries if the peers are removed
- we need to make sure peers defined in the server config gets passed to the resulting configuration installed on the server

codedbydro, 2026-09-16:
I've been looking through the current WireGuard/IPAM implementation for understand the scope of #788.

I think the previous discussion proposed a dedicated VpnPeer model, but the scope was left open. After tracing the current implementation, I think there are a few questions that are still important before choosing an implementation:

1. Should manually configured peers remain part of the existing Vpn.config, or should OpenWISP introduce a separate representation for them?
2. Is the peer's tunnel IP need to be explicitly provided by the user, or should OpenWISP allocate it from the VPN subnet?
3. If the peer is defined in the VPN configuration, what should be considered the source of truth for its IPAM reservation the peer's allowed_ips or a separate IPAM allocation?
4. How should the lifecycle of that reservation be tracked so removing a peer does not accidentally release an address belonging to a VpnClient or another IPAM object?
5. Is this intended specifically for WireGuard server peers, or should the implementation also cover the existing VXLAN-over-WireGuard peer handling?

From the current code, I can see that _add_wireguard() currently replaces the configured peers with the peers generated from VpnClient, while IP allocation is currently tied to VpnClient through IPAM. So these questions seem to be the main architectural gap between the requested behavior and the current implementation.

I'm not proposing an implementation yet; I wanted to share what I found and understand the intended scope before going further.

codedbydro, 2026-09-16:
I've been looking through the current WireGuard/IPAM implementation to understand the scope of #788.
After reviewing previous discussion around introducing a dedicated VpnPeer model, but since the scope was left open, I traced the current implementation further before proposing a solution. One important thing I found is that the existing WireGuard configuration editor already exposes peers in Vpn.config, and the API also allows them to be written. However, during configuration generation, _add_wireguard() currently replaces the configured peers with the peers generated from VpnClient, so manually configured peers are effectively discarded.
On the IPAM side, IP allocation is currently tied to VpnClient, and I couldn't find an existing convention for managing the lifecycle of an IpAddress reservation belonging directly to an entry in a JSON peer list.
This leaves a few architectural questions that seem important to clarify before implementing:

1. Should manually configured peers remain part of the existing Vpn.config, with their IPAM reservations reconciled separately, or should OpenWISP introduce a persistent representation for these peers?
2. Should the peer's tunnel IP need to be explicitly provided by the user, allocated automatically from the VPN subnet, or support both?
3. If peers remain in Vpn.config, should their allowed_ips be the source for the IPAM reservation, or should the reservation have a separate explicit IP?
4. How should the reservation lifecycle be tracked so removing a peer cannot accidentally release an address belonging to a VpnClient or another IPAM object?
5. Is this intended specifically for WireGuard server peers, or should the implementation also cover the existing VXLAN-over-WireGuard peer handling?

I'm not proposing an implementation yet. I wanted to share what I found in the current code and confirm the intended scope/architecture before going further.

KimiNoLie, 2026-10-16:
Hi @codedbydro
! I’m interested in working on this issue and saw your earlier investigation. Are you still planning to work on it, or would it be okay for me to pick it up? I’d like to coordinate before starting to avoid duplicating your work.