openwisp/openwisp-controller #793: [bug] Unable to import devices from CSV

pandafy, 2023-07-14:
**Steps to replicate**

1. Export devices using the export button on device changelist to CSV format
2. Import the exported devices in the previous step

**Actual Result**
[image]

**Expected result**
Device are imported without any errors

pandafy, 2023-07-14:
My browser's UI uses dark mode, hence CSS defined in django-import-export is using dark mode colours. But, this breaks the theme of OpenWISP. 

[image]

pandafy, 2023-07-14:
We will need to override these rules
https://github.com/django-import-export/django-import-export/blob/4ce494ac5c30df85bb5b1ad5d436ef64fd72d35e/import_export/static/import_export/import.css#L83-L115

nemesifier, 2024-04-09:
I split the dark mode issue here: https://github.com/openwisp/openwisp-controller/issues/851.

commit 688057b0 (2024-04-11): [fix] Fix import errors from files generated by OW #793

Ensured the following works:

- Export devices using the export button on device changelist to CSV format
- Import the exported devices in the previous step

Added tests for main use cases.
Some cases may be missing and we may
have to fix them as we encounter them.

Closes #793
commit 912297f0 (2024-04-11): [fix] Fix import errors from files generated by OW #793

Ensured the following works:

- Export devices using the export button on device changelist to CSV format
- Import the exported devices in the previous step

Added tests for main use cases.
Some cases may be missing and we may
have to fix them as we encounter them.

Closes #793
commit 1e641fe0 (2024-04-17): [fix] Fix import errors from files generated by OW #793

Ensured the following works:

- Export devices using the export button on device changelist to CSV format
- Import the exported devices in the previous step

Added tests for main use cases.
Some cases may be missing and we may
have to fix them as we encounter them.

Closes #793
PR #852 (2024-04-20): [change/fix] Improved device export #850
- Added more device, config and geo fields to export
- Fixed UUID bug #850
- Fixed import bug #793 

Fixes #793
Fixes #850
closed 2024-04-20 by nemesifier