openwisp/openwisp-controller #808: [feature] Status message

mips171, 2023-08-25:
Would be great if we could set a status banner or other type of message that all users see when they log in, so we can communicate with them via the system. Stuff like "new firmware available" or "system is experiencing an issue right now".

okraits, 2023-08-26:
What about a notification? Is it not prominent enough?

nemesifier, 2023-08-29:
This could be implemented by overriding the base templates without even touching OpenWISP.
See https://docs.djangoproject.com/en/4.2/howto/overriding-templates/ .
I don't see a very compelling reason to spend resources on this.

mips171, 2023-09-06:
Yeah I think a notification wouldn't be immediate enough. For example, someone who logged in but didn't check their email recently, or the notification is buried in 99+ others about device statuses. An acknowledgeable/closable banner at the top of the screen seems like the way to go to me.

@nemesifier that should be workable. But I still feel it would be cool to have it as an option in the frontend. Maybe there's already a good Django plugin to add the feature and allow editing the message text from the admin GUI?